selfcustody/krux #2: Use PSBTView for memory efficient PSBT handling

jreesun, 2021-07-25:


jreesun, 2022-03-01:
This was a bigger problem when the firmware was larger and the full SRAM wasn't being taken advantage of, so I'm going to close this issue for now until (if?) it starts to becomes a problem again.

jreesun, 2023-08-18:
[See here for more discussion](https://github.com/selfcustody/krux/discussions/229)

odudex, 2024-05-27:
We have made several improvements to RAM management, allowing for the signing of larger transactions. For instance, transactions with up to 100 inputs can now be signed using QR codes, and those with up to 145 inputs can be signed using SD cards.
Additionally, we have optimized flash management, reducing its usage frequency and making its content smaller and more predictable. At this time, we prefer to avoid using PSBTView, as it would significantly increase flash usage and leave behind large amounts of data.
Yet, PSBTView remains as a ace up one's sleeve in case we need to sign larger transactions.

tadeubas, 2024-06-17:
Could we increase the heap size a little again? I think the transaction used was from a single-sig wallet, IMHO it is necessary to pass all [lopp tests on HW signing performance](https://blog.casa.io/bitcoin-multisig-hardware-signing - performance-2024/) and be able to withstand at least 50% more, what do you think?