selfcustody/krux #12: Maix Amigo support

stepansnigirev, 2021-08-02:
Hi! Great project!
Do you think it might be possible to make krux compatible with Maix Amigo?
It is a pretty cheap devboard that has a large screen with touch, camera and SD card slot.
MCU is the same as in M5StickV - K210.
https://www.seeedstudio.com/Sipeed-Maix-Amigo-p-4689.html

jreesun, 2021-08-02:
Interesting! Yes, that should be possible. It _probably_ wouldn't take a ton of effort to add support in the firmware for it, however, the experience on that device would likely be subpar without additional effort. 

Krux's UI / input logic right now is unfortunately tightly coupled with the M5StickV and wouldn't take advantage of the extra screen space in a smart way, and it definitely wouldn't support touch input/buttons without some work. The build process would also need to now take the device into account when deciding which 'project' to build the firmware for, etc.

I'll need to get my hands on one of these devices and see what I can do.

valerio-vaccaro, 2021-08-03:
Nice project, 
   patched maixpy firmware is running on amigo (need some fix on cmake flags like for m5stickv and same builtin python files)
Some fix needed in buttons definition/screen/etc, menu crash with an "out of memory alert" but i think we can reduce the display allocated buffer.
In next days i think i will create a branch for amigo boards.

jreesun, 2021-08-16:
@valerio-vaccaro I have an Amigo now to test with if you're still planning on putting up a PR! Hope the recent changes in `main` don't make this too much of a headache 😅 

Until then, I'm focused on [this issue](https://github.com/jreesun/krux/issues/9) and will circle back to this after that's complete

closed 2022-04-23 by jreesun