selfcustody/krux #18: Testing on Manjaro OS vs MacOS

kkdao, 2021-09-13:
I tried to get through the commands on the Manjaro OS and it was fine for building and flashing, but with the microSD section and the `./krux flash-software /Volumes/SD` command, it does not copy over the necessary files properly. The contents are the same as what is in the Krux directory.

When I try with MacOS it copies over the necessary python files and directories, and this time when the microSD is loaded into the device it boots successfully.

I have not tested with more popular linux OS's like Ubuntu, PopOS, etc., but hopefully it's not an issue there as well otherwise that might be deterring folks from getting started with Krux because some who uses a linux OS mainly may not have a Mac handy.

jreesun, 2021-09-15:
Oof, good to know! I think I see the problem you ran into. I'll put up a fix shortly; thanks for the report!

jreesun, 2021-09-15:
@kkdao Does this fix resolve the issue you ran into?

closed 2021-09-16 by jreesun