selfcustody/krux #48: Vietnamese translation

jreesun, 2021-11-02:


AshleyFires, 2021-11-05:
Can you provide more information about this please?

jreesun, 2021-11-05:
Sure! Krux now has limited unicode support (latin and extended latin characters with diacritics), so I'm asking for help with translating its UI to the various languages that can be rendered with those characters. I _think_ this would include Vietnamese?

To add a translation, all you need to do is create a new file in the `i18n/translations` directory named after the locale that corresponds to it (`vi-VN.json` in this case). In the file, you map the English versions of the phrases to their Vietnamese equivalents.

Does that answer your question?

AshleyFires, 2021-11-05:
Not sure I can translate it 100% right but I think I can help you a little bit about this issue. Can I join this project with you? 👍

jreesun, 2021-11-05:
Yep, just make a pull request with your changes and I'll look it over. Thanks!

AshleyFires, 2021-11-06:
Thanks, I'll try my best!

jreesun, 2021-11-25:
Resolved by #64

closed 2021-11-25 by jreesun