selfcustody/krux #78: MaixPy should be a submodule

Giszmo, 2022-02-08:
Copying code in makes code review hard as now the reviewer also has check where that copied code came from and if it was modified. #39 sounds like this issue might be resolved by removal of the code but anyway, for a Bitcoin wallet code auditability is key.

jreesun, 2022-02-09:
The changes made to MaixPy are extensive enough (and specific to Krux, such as memory wiping and QR code generation) that in order to do a proper audit, you do need to review the firmware code regardless. 

I think making it a submodule eventually would be worthwhile (if/when Krux is more platform-agnostic), but for now since the firmware code is Krux-specific there isn't much reason to break it out yet. I'll leave this issue open though since it makes sense to do that in the future.

jreesun, 2022-02-10:
Okay, I see. I can certainly try to do that to make auditing easier 👍

closed 2022-03-21 by jreesun

jreesun, 2022-03-21:
@Giszmo Hey, I just wanted to give you notice that I'm planning to put out the first official release (v1.0.0) of Krux this week!

PR #102 (2022-03-21): Add MaixPy as submodule
Resolves #78 and partially addresses #101

I forked https://github.com/sipeed/MaixPy from commit https://github.com/sipeed/MaixPy/commit/8972141004dd03aa13346185c0c0d6d8d90874e6 and made [my fork](https://github.com/jreesun/MaixPy) a submodule at `firmware/MaixPy` in Krux now.

As https://github.com/jreesun/krux/issues/78#issuecomment-1034315370, I consolidated all my changes into one commit to make an audit easier:
https://github.com/jreesun/MaixPy/commit/90b61f71f013341077eb976d7f43246c2feeddc2

As part of that work, I decided to minimize my changes to sipeed's versio [... cut]