selfcustody/krux #83: xpubs, legacy addreses and BlueWallet

odudex, 2022-02-18:
After a few code tweaks I succeeded in making Krux export singlesig wallets to Blue and Sparrow wallets as well sign and export PSBTs back to them. [Video](https://twitter.com/odudex/status/1493579745761665024?s=20&t=NbTRzA_jz4kszKoJfnR4rQ)
However, there still an issue with Blue:
If I use a zpub everything works fine, but if I import the xpub in it, it will automatically create a weird legacy wallet with m/84'/0'/0' (native segwit) derivation.
Since xpub are commonly used for legacy, and if I understood the Krux code, it is incompatible with legacy addresses, Krux shouldn't offer xpus, only zpubs. What do you think?
(I'm new to all of this)

jreesun, 2022-02-20:
This is a good question and something I should add to the docs to clear up potential confusion about what Krux exports.

Krux treats xpubs the same way that [Specter Desktop does](https://github.com/cryptoadvance/specter-desktop/blob/master/docs/faq.md#why-when-i-export-a-multisig-wallet-from-specter-desktop-settings--export--copy-wallet-data-created-from-devices-with-only-segwit-zpubs-do-i-get-a-data-structure-with-expected-segwit-derivation-paths-but-xpubs-instead).

The xpub that Krux exports in the QR code follows the [bitcoin core descriptors spec](https://github.com/bitcoin/bitcoin/blob/master/doc/descriptors.md#key-origin-identification) and includes key origin and derivation info in brackets that makes unofficial standards like zpub and ypub unnecessary _if the wallet software is able to parse this additional information_. From the spec:
> Every public key can be prefixed by an 8-character hexadecimal fingerprint plus optional derivation steps (hardened and unhardened) surrounded by brackets, identifying the master and derivation path the key or xpub that follows was derived with.

In practice though, not all wallet software supports this format, so Krux provides a zpub as a fallback.

Unfortunately, due to screen size constraints on the M5, this additional information in brackets can't be shown to the user, so _only_ the 'classic' base58 xpub representation is shown to the user in plaintext.

To give a few examples, let's say you used the same mnemonic to generate a single-key and a multisig wallet key in Krux: `olympic term tissue route sense program under choose bean emerge velvet absurd`

The single-key xpub shown to the user would be:
```
xpub6DPMTPxGMqdtzMwpqT1dDQaVdyaEppEm2qYSaJ7ANsuES7HkNzrXJst1Ed8D7NAnijUdgSDUFgph1oj5LKKAD5gyxWNhNP2AuDqaKYqzphA
```

However, the QR code would output:
```
[55f8fc5d/84h/0h/0h]xpub6DPMTPxGMqdtzMwpqT1dDQaVdyaEppEm2qYSaJ7ANsuES7HkNzrXJst1Ed8D7NAnijUdgSDUFgph1oj5LKKAD5gyxWNhNP2AuDqaKYqzphA
```

Similarly for multisig, the user would see:
```
xpub6EKmKYGYc1WY6t9d3d9SksR8keSaPZbFa6tqsGiH4xVxx8d2YyxSX7WG6yXEX3CmG54dPCxaapDw1XsjwCmfoqP7tbsAeqMVfKvqSAu4ndy
```

But the QR code would output:
```
[55f8fc5d/48h/0h/0h/2h]xpub6EKmKYGYc1WY6t9d3d9SksR8keSaPZbFa6tqsGiH4xVxx8d2YyxSX7WG6yXEX3CmG54dPCxaapDw1XsjwCmfoqP7tbsAeqMVfKvqSAu4ndy
```

Having said all that, I may separate the xpub text from the QR code and display them on two back-to-back screens rather than putting them on one so that I can display the bracketed information to the user. No surprises, WYSIWYG.

jreesun, 2022-02-20:
You mentioned that code tweaks were necessary to make things work, so it sounds like there is a bug somewhere... Could you link the changes you had to make?

odudex, 2022-02-20:
Thanks for the explanation about descriptors!
I was getting an exception on psbt.py
Added a "list" convertion on:
If len(list(out.bip32_derivations.values())) > 0:
Than it worked. You can check all the code on my tests brach

jreesun, 2022-02-21:
Great catch! Must be a discrepancy between Python 3 and (Maixpy) Micropython since the tests didn't hit that. Fun!

jreesun, 2022-02-24:
@odudex I added your fix to the `tests` branch and just merged it into `main` now. Does that resolve this issue?

odudex, 2022-02-24:
@jreesun I think so. If I have some time I'll test a fresh build from `main` to make sure.

closed 2022-02-24 by odudex