selfcustody/krux #114: Font which width is not a byte multiple won't render

odudex, 2022-03-29:
When fonts are baked according to instructions they will render correctly only if its width is a 8 multiple. Ex 8*14, 8*16, 16*32.
Other sizes like 12*24 won't render, because 12 can't be divided by 8. However, if I lie to `py_lcd.c` on config, and say it is a 16*24 font, it will render, with larger space between letters thought.

odudex, 2022-04-08:
Had some more spare time and solved this issue, all fonts are rendering correctly now [ex: 14x28](https://photos.app.goo.gl/BX6tDcLNE6oziDfDA).
Before a PR, what do you think of a cleanup on font.c and remove those unused features (monospace, scale, x_spacing, y_spacing)

jreesun, 2022-04-08:
That sounds good to me 👍

odudex, 2022-04-09:
Done, removed those variables and also all `float` variables and methods (round, ceil, etc) from this part of the code. Also added proper PMU control for Amigo

closed 2022-04-23 by jreesun