selfcustody/krux #122: Fully reproducible releases

jreesun, 2022-04-20:
Explore making releases fully reproducible such that someone could checkout the tagged commit of a release, manually build Krux, and get the same sha256 as the one available for download from the releases page.

Building everything inside a Dockerfile already brings us pretty close, but it's possible for example that the apt packages being installed in `build-base` could change versions from one run to the next, etc. What we want is for the `build-base` image to be the same for everyone, while still itself being reproducible.

We _could_ just have GitHub, in a workflow script, build and publish the `build-base` image, and update the Dockerfile to pull this image down instead. That would solve the problem of reproducibility in the sense that hashes would match, but it would also mean that we have to trust GitHub's CI when building the image.

The ideal solution would be to have a way to reproducibly build the `build-base` image from scratch locally and have it match, exactly, the one being automatically generated by GitHub's CI. This will involve being precise with all apt packages (making sure _their_ hashes match, not just version number)...

dockpin seems like a potential solution to this problem.

tadeubas, 2024-03-14:
Can we consider this done in [release 24.03.0](https://github.com/selfcustody/krux/releases/tag/v24.03.0) ?

odudex, 2024-03-14:
Yes, unless someone proves the opposite, it was done in https://github.com/selfcustody/krux/releases/tag/v24.03.0

closed 2024-03-14 by odudex