selfcustody/krux #127: Error on flashing binaries from Macs

odudex, 2022-04-26:
MacBook:krux-v22.03.0 xxx$ ./ktool-mac -B goE -b 1500000 maixpy_m5stickv/kboot.kfpkg
[10684] Error loading Python lib '/var/folders/pv/58vvqdhn26q3fvsnbqq6pjb00000gn/T/_MEIGLO3yx/libpython3.9.dylib': dlopen: dlopen(/var/folders/pv/58vvqdhn26q3fvsnbqq6pjb00000gn/T/_MEIGLO3yx/libpython3.9.dylib, 10): Symbol not found: ____chkstk_darwin
  Referenced from: /var/folders/pv/58vvqdhn26q3fvsnbqq6pjb00000gn/T/_MEIGLO3yx/libintl.8.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/pv/58vvqdhn26q3fvsnbqq6pjb00000gn/T/_MEIGLO3yx/libintl.8.dylib

As a workaround I suggested to use kflashGUI instead

jreesun, 2022-05-06:
@odudex Did you ever get more information about this (e.g., if their Mac OS version was <11 or >11)? 

To resolve this, I was thinking I could update the Github workflow in `ktool` to use the oldest available Mac OS version, and also include `ktool.py` in the packaged release as a fallback that the user can run under their own python installation.

odudex, 2022-05-06:
One case version was 10.13.6

jreesun, 2022-08-01:
@odudex Going to close for now since I just put out a new release with builds for Mac OS 10 that should hopefully address this user's issue. The new ktool builds will make their way into the 2022.08.0 release.

closed 2022-08-01 by jreesun