selfcustody/krux #129: Add BIP39 Passphrase

odudex, 2022-05-02:
Requested feature

jreesun, 2022-05-03:
AFAIK, a passphrase can be any UTF-8 string. That said, most people probably only use the keys found on their keyboard, `a-zA-Z0-9` with special characters (`@#$` etc.). Unfortunately, these keys won't all fit on the screen. To support this, we would need to add a new keyboard input method that functions similar to how mobile devices do it: default to lowercase letters with a Shift button to make them all-caps until a letter is pressed, and a "123" button to switch to numbers (and "abc" to go back to letters).

Since a passphrase is used as an additional layer of security and/or for plausible deniability, I think by default we shouldn't prompt for it after entering the mnemonic. Instead, we could add a new boolean `prompt_for_passphrase` setting under `Settings` that, only when enabled, asks the user if they want to enter a passphrase after loading their mnemonic (still giving them the option to leave it blank). This way, the existing load behavior can remain unencumbered and (relatively) fast to get through, and people using passphrases can obscure this fact (or not, if they decide to persist the setting on a microSD card).

odudex, 2022-05-22:
Have passphrase working.
Created "long press" for page button to toggle between pads
[Keypad](https://photos.app.goo.gl/9MS9b98Xf5VbJppTA)
Still have to create the setting to prompt or not. Thinking about ways to change keypads easier with touch in Amigo

odudex, 2022-05-24:
Passphrase working on all devices, [swipe gestures](https://github.com/diybitcoinhardware/embit/issues/28) implemented in Amigo to switch keypads.
Still to do: Respective setting, tests, simulations and documentation.
Still didn't got how to run simulator neatly.
Posted the code on my [passphrase branch](https://github.com/odudex/krux/tree/passphrase), will wait previous PR to be merged before make new ones

jreesun, 2022-07-16:
@odudex What all needs to be done to get this into a state that could be reviewed and merged? I had planned to make some follow-up changes to the code in `develop` after I merged your last PR, but I see that your `passphrase` branch has a lot of follow-up changes now as well, including a `Pad` class that I'd like to see incorporated before cutting a new release.

odudex, 2022-07-16:
With passphrase I had a "full green" code on tests and coverage. Than I added rotary encoder, Dock, etc. Since you're back I'll stop adding things, fix the tests and make a PR. But, as everything is working, if you prefer I could make the PR the way it is.

jreesun, 2022-07-16:
I started working on the simulator to add support for Amigo (not going to add the Dock or Bit right now since they're more custom) and will be updating the documentation to be less focused on the M5StickV and include simulator screenshots of the Amigo too; if you want to polish your PR and get the tests passing while I do that, that would be great!

Thanks for all your work on this

odudex, 2022-07-16:
Sounds a good plan!
I will make a green PR next week!

closed 2022-07-20 by jreesun