selfcustody/krux #139: Add i18n support to online documentation

qlrd, 2022-07-16:
Hi,

In a [fork](https://github.com/qlrd/krux/tree/docs-translation) I created some support for `i18n` of the online documentation.

A preview of this multilingual documentation is available [here](https://qlrd.github.io/krux/), as well as I created a [README](https://github.com/qlrd/krux/blob/docs-translation/i18n/README.md) for anyone who wants to help.

To change the documentation language, you can click on a translation icon to the left of the search field. For now, support for Spanish, Esperanto and Portuguese.

(Note: some translations were done automatically, just as a means to make it possible)

I believe that as soon as a minimum of acceptable translation is possible, and with Jeff's approval, it is possible to open a PR.

jreesun, 2022-07-16:
Awesome!! Great work, thank you.

Could you make a PR where you add support for i18n without translations (except for English)? I plan to make a fair number of changes to the site to make it less M5StickV-focused and include Amigo information, so I'll probably be changing the text in a lot of places and don't want to make you do unnecessary translations (again 😅 ). So, ideally, the order of operations here would be:
1. You make your i18n PR that lays the foundation for translations to be added
2. I make changes to the English site in my PR
3. Then you (or others) can add translations for the new site in follow-up PR(s), ideally one PR per language

Does that work for you?

qlrd, 2022-07-17:
I think it works, lets try.

Only to be clear:

I dint change any of the English files, only changed their names, so the `mkdocs-static-i18n` can find the properly translation. Any of the original contents are followed by an `en` (or `es`, `se` or `pt`) and the extension `md`.

Example:

1. `docs/index.md` are now:
  - `docs/index.en.md` (English)
  -  `docs/index.es.md` (Spanish)
  -  `docs/index.se.md` (Esperanto)
  -  `docs/index.pt.md` (Portuguese)
  
2. `docs/getting-started/generating-a-mnemonic.md` are now:
  - `docs/getting-started/generating-a-mnemonic.en.md` (English)
  -  `docs/getting-started/generating-a-mnemonic.es.md` (Spanish)
  -  `docs/getting-started/generating-a-mnemonic.se.md` (Esperanto)
  -  `docs/getting-started/generating-a-mnemonic.pt.md` (Portuguese)

Maybe you can add the planned new content on `*.en.md` files?

jreesun, 2022-07-17:
> You will add the planned new content on *.en.md files?
> Your idea is to remove es, se and pt files and leave only the en files?

Yes, that is correct. I would like a PR containing only the new `.en.md` files and the other changes required for i18n on the site to work. If you want to include the `.es`, `.se`, and `.pt` files you already created in the PR, that is fine too. I was just requesting that you wait until the new text is available before adding more translations so your time won't be wasted if the English versions of the text change.

closed 2022-08-01 by jreesun