selfcustody/krux #146: Makefile to make development easier and faster

qlrd, 2022-07-22:
Some **development** tasks require, on an extended time scale, a considerable amount of repetition of the commands. Example for updates, format, lint, test, simulate, build and flash firmware: 

1. `git pull origin <branch> && git submodule update --init --recursive`
2. `poetry update`
3. `poetry run black ./src && poetry run black ./tests`
4. `poetry run pytest --cache-clear --cov src/krux --cov-report html ./tests`
  4.1. verbose: `poetry run pytest --cache-clear --cov src/krux --cov-report html --show-capture all --capture tee-sys -r A ./tests`
6. `cd simulator && ./run-all-sequences`
7. `vagrant up`
8.` vagrant ssh -c 'cd /vagrant; ./krux generate-keypair'`
9. `vagrant ssh -c 'cd /vagrant; ./krux pem-to-pubkey pubkey.pem'`
10. `vagrant ssh -c 'cd /vagrant; ./krux build maixpy_m5stickv'`
11. `vagrant ssh -c 'cd /vagrant; ./krux flash'`

I propose to simplify, for the developer, the repetition of these commands in some leaner ones, through a `Makefile`. The example above would then be:

* 1 and 2: `make update`
* 3: `make format`
* 4 and 4.1: `make test` and `make test/verbose`
* 5: `make simulator/<some device>`, where `<some device>` could be a supported one:
* 8, 9 and 10: ` make build/sign`
* 8 and 10: `make build/<some device>`
* 11: `make flash/<some device>`

I made an example [here](https://gist.github.com/qlrd/95d2a6431b2fd854c1d658395c763966). 

For more details, see this [discussion](https://github.com/selfcustody/krux/discussions/144)

tadeubas, 2024-04-02:
I think with the addition of `poetry run poe` tasks this problem is "solved", right? I've created a few tasks that I'm using to help with development, and I think anyone could add new tasks to suit their needs.

Some examples (all commands start with `poetry run poe`):

- format 
- lint
- test-simple
- i18n-build
- simulator --sd
- git-update

All current tasks are listed here: https://github.com/selfcustody/krux/blob/develop/pyproject.toml#L64

odudex, 2024-04-02:
Agreed! We could close this @qlrd

closed 2024-04-02 by qlrd