selfcustody/krux #149: Add printer support to Amigo

jreesun, 2022-08-02:
Currently, Krux can only print QR codes via thermal printer with the M5StickV. 

The Amigo should be able to do this too since it has grove connector ports.

jreesun, 2022-09-17:
@odudex Did you have any luck with this?

odudex, 2022-09-17:
Sorry, I did not even buy the printer yet. There's a guy in the Telegram group that succeed with another M5STICKV, and others have ordered printers

jreesun, 2022-12-04:
@odudex You mentioned that you got it working on Amigo. Is this working now with the latest code in `develop`, or are more changes needed before I can close this?

odudex, 2022-12-05:
Ad I remember, it worked out of the box. I just changed some timing values to get more contrast on my AliExpress printer. For my latest test firmware I also expanded the prints for text, compact SeedQR and Tiny Seed (which is now scannable too, as the real thing). Maybe it's time to clean this new code and make a new PR, I also didn't do pytest for those too😬.

closed 2023-02-18 by jreesun