selfcustody/krux #150: Add support for Nunchuk

jreesun, 2022-08-07:
Users who have tested Krux with Nunchuk haven't been able to make it work. Some time needs to be spent looking into how Nunchuk handles QR codes and what encoding it expects xpubs and PSBTs to be in.

kkdao, 2022-08-15:
The problems I faced getting it to work with Nunchuk had to do with the devices having trouble scanning the presented QRs when exporting PSBTs. With the m5stickV, you have to hold the device close up to the animated QR and then it takes a long time to 100% scan it. With the Amigo, you have to hold the device so that the display surrounds the QR code and then it loads/scans OK. 
 With Nunchuk I'm selecting the "Export to Keystone/Seedsigner" option (other option is Passport), whereas with Sparrow it just has "Show QR" which presents a large animated option when exporting PSBT. That's likely why it's easier to use Sparrow for exporting because the QRs displayed are larger.

Note: no option to load wallet into Krux with Nunchuk compared to Sparrow (use term export), so had to load wallet into Krux using Sparrow before I sent a txn with Nunchuk.

odudex, 2022-09-26:
I successfully used Krux and Nunchuk for Android to sign single-sig PSBTs. With current Nunchuk available on Play Store the xpub scan is not working, so it's necessary to scan it in another app, such as G. Lens, than paste it in the Nunchuk's descriptor text field.
Good news is next Nunchuk mobile release will be compatible with Krux out of the box. They added the missing "raw QR code scan" feature and sent an .apk to test. It's great!

closed 2023-02-18 by jreesun