selfcustody/krux #165: 99 D6 dice rolls instead of the current 100

qecez, 2022-08-13:
Both SeedSigner and Coldcard take in 99 D6 dice rolls for the 24 word mnemonic. 24 words give a full byte of checksum and it would be nice to have it consistent across different signers.

jreesun, 2022-08-14:
I see why this is wanted, but I want to note a few things:
1) ColdCard says "**At least** 99 rolls are required for 256-bit security", which implies the user can roll 100 times (or more) if they want to. So, a 100-roll string from Krux should still be verifiable with ColdCard's [rolls script](https://coldcard.com/docs/rolls.py).
2) Technically, 99 * log2(6) ~= 255.911287571, which is very close to but not quite 256 bits. 100 * log2(6) gives you ~258.496250072 bits. At least [one other person on the internet agrees with me](https://www.reddit.com/r/coldcard/comments/c99457/99_or_100_dice_699_2256_6100/)!
3) The original decision to go with 100 over 99 was intentional because I felt it was more obvious that doubling the rolls would double the entropy, plus `(50, 100)` and `(30, 60)` are "cleaner" sets of numbers to remember.

I'm fine with standardization when it results in something better and more portable (for users), but in this case I'm on the fence because we're talking about key generation which is a one-time operation and we would technically be moving to something less secure in the name of standardization.

That said, I'm aware this is a somewhat pedantic argument due to the miniscule difference in bits we're talking about, so I'm still open to other opinions. I just don't think this change should be rushed until it's clear what the community prefers.

Maybe I'll leave it up to a vote and give it a month: thumbs up for switching, thumbs down for leaving it as-is.

jreesun, 2022-09-02:
@qecez Going to close this in favor of https://github.com/selfcustody/krux/issues/174 

I think it resolves the issue for both of us. Check it out and let me know what you think!

closed 2022-09-02 by jreesun