selfcustody/krux #171: Test and enable GRBL cnc "printer" implementation

jreesun, 2022-08-19:
We need a brave person with a CNC machine (powered by a GRBL controller) who would be willing to test the `GRBLPrinter` implementation here:
https://github.com/selfcustody/krux/blob/develop/src/krux/printers/cnc.py#L326-L376

It uses the same underlying `GCodeGenerator` base class that the `FilePrinter` uses (which we know to operate correctly); it's just a matter of confirming that the GRBL-specific serial communication logic functions as expected--or not, in which case changes to make it function correctly would be greatly appreciated!

InfiniteQE, 2023-06-27:
I've started testing the CNC features of Krux from odudex's release of v23.04.beta11

I am currently doing some testing with [UGS](https://winder.github.io/ugs_website/) and will update over the next few days with any other issues or bugs that arise when I start milling via MicroSD card and [offline controller](https://www.foxalien.com/products/offline-controller-module-for-cnc-router-masuter-series-wm-3020) (no PC used going from Krux -> airgapped CNC)

- 1 : When Krux is writing the Maix Amigo's MicroSD (30+ seconds), the device appears frozen on "Checking for SD card" the whole time, it should update the screen text to "Please wait ...."

- 2 : The cnc.py hard codes the output as "qr.nc" and will overwrite an existing file.  Instead it would perhaps be better if the different QR codes were given unique names like "PlaintextQR.nc", "CompactSeedQR.nc", "SeedQR.nc" and if the file already exists, to append a number SeedQR-001.nc, etc

- 3 : I would also request the gcode output of plaintext. One can enter arbitrary text/numbers/special characters via the Tools menu to output as a QR code, but I would like to output to gcode the mnemonic seed words in plain text, as well as arbitrary text/digits/special characters to gcode.  
This would allow a large number of possibilities from engraving [mnemonic seed words on metal hexagonal rods](https://coldbit.com/product/coldbit-hex/), or [arbitrary text](https://www.printables.com/model/75703-hodl-hex-wallet-0375-hex-rod-stamp-jig), or [a passphrase](https://coldbit.com/product/coldbit-passphrase/).  What I am really keen on is using gcode to output shamir secret sharing shards of the mnemonic - again [Coldbit has the same idea](https://coldbit.com/product/coldbit-hex-5x-bundle/) but using an airgapped stateless signer to mill the words into steel/titanium will save hours of hammering.

- 4 : I would love to see Krux able to import either plaintext via MicroSD or import plain text via QR code (or maybe import via animated QR code) to then convert to gcode.  This would open up the possibility of [CNC engraving an impossible-to-type multisig backup](https://bitcoiner.guide/multisig/backup/#public-key-backup), such as what Specter can save to PDF - Advanced backup details with all of the multisig quorum wallet descriptors, fingerprints, derivation paths and cosigner's xpubs.

jreesun, 2023-06-28:
Awesome! Glad to see someone finally try this out :) For the record, there are two CNC "printers" here: one that writes the gcode to a microSD and another that streams the instructions to a GRBL controller. The latter is what I was not able to test and am curious how it works. Because I couldn't test it, I commented it out. It would need to be uncommented first and a new build created so you could select this option. Could @odudex do this for testing purposes?

2: 👍 I think it should use the same file browser logic as we use for signed PSBTs being saved to microSD.

3: Agree that this would be useful. Can you open a new discussion about this feature specifically? There's probably a lot to hash out here about how this could be done, and what all should be customizable. Also need to keep in mind that the CNC "printer" operates on the same interface as the thermal printer. The original intention behind this was that a "Printer" would have a common set of things that could be printed out (just QR codes to start), and each implementation of printer would output them in their own way.

4: 👍 I think this should be an option when viewing your multisig to print the backup in textual, wallet descriptor form (rather than as a QR code), and use the CNC printer method if selected.

InfiniteQE, 2023-06-28:
@jreesun I thought I would share here first my CNC test, this was a new mnemonic generated by camera entropy, saved to gcode as a plain text QR on MicroSD then milled with the offline controller.

The only modification I made the the Krux Settings/Printer/CNC was to change the work size to 2" from the default 3.5".

On the offline controller I manually jogged the XY axis, set zero approx 1" from the lower left corner of the workpiece, the Z just above, and had to manually start the spindle motor with the offline controller (as Krux's gcode does not start or stop the spindle).

I then loaded the qr.nc file from Krux's MicroSD, and ran the milling operation.

I was using a non fluted vbit that was included with my CNC and a piece of MDF.  I used my shopvac to remove the sawdust before taking these photos.

[image]
[image]

Impressions :
- The gcode does work, I inspected a few samples earlier this week with Universal Gcode Sender and the Krux screen displayed QR matches the gcode QR
- Krux by default stores the CNC printer's settings parameters (work size, feed rate, etc) by default on Krux's flash, there is no way for users to delete these stored values to restore defaults.
- Krux defaults to imperial values, there is a setting to change from inches to mm, but this doesn't change the saved values which remain in inches (resulting in a QR code output of 3.4mm)
- Krux does not display the units for CNC settings, some are obvious like work size / borders, but others like plunge rate and feed rate are ambiguous.
- it would be nice to be able to reset or restore to defaults the Amigo's flash stored values (I tried reflashing the firmware which did not restore defaults)
- Do you have a suggestion for a cable to connect Maix Amigo to a grbl CNC control board ? My offline controller has ribbon cable with 8 pin connectors on either end, same approx shape as Amigo, but the connector doesn't fit.
- Krux doesn't start/stop the spindle motor, unlike the gcode samples included with the offline controller.
- Krux might have an alignment issue with where it places the QR code, I initially zeroed the XY axis at the bottom left edge corner of the workpiece, but I noticed on my first test, the top line of the QR code (engraved from right to left) overshot and went negative of my X zero point, so I spun the wood around and re zeroed about an inch off the lower left corner in XY, which was a succesd
- Thank you for your thoughts on Multisig and arbitrary text engraving, I'll start a new github issue