selfcustody/krux #176: "Auto Go" + Erase words

odudex, 2022-09-02:
Also feedback from Telegram:
Keypad's "Go" could be automatically called after 4 numbers are typed when entering seed "as numbers".
To be able to correct mistakes, when text box is empty, the del "<" key could erase the last typed seed word (not only in numbers keypad).

odudex, 2022-09-06:
Implemented shutdown confirmation, auto-Go for "Via Text" and "Via Numbers" and word deletion for all types of inputs, that could save a lot of clicks/touches. [video](https://photos.app.goo.gl/Y5vQ5ptJEvYA3Z6c9)
But before if refactor tests and make a PR:
@jreesun What do you think about this?

jreesun, 2022-09-07:
I think the "auto go" part is good, and I like the confirmation when you choose to shutdown. But the `<` key being used to delete the previous word feels odd to me. Could we instead have the screen that shows the word you've selected (which currently requires a button press/tap regardless) have a Yes/No prompt before proceeding? If you select No, it drops you back at the input screen.

odudex, 2022-09-07:
Will do it the way you suggested. It is simpler and more intuitive.

closed 2022-12-04 by jreesun