selfcustody/krux #177: "Cannot sign" PSBTs from Blue and Electrum

odudex, 2022-09-12:
Warned by a Telegram user, I verified signing PSBTs transactions from wallets generated by zpubs, like in Blue and Electrum mobile, is not working.
This error is being raised:
```
sigs_added = self.psbt.sign_with(self.wallet.key.root)
        if sigs_added == 0:
            raise ValueError("cannot sign")
```
PSBTs from Sparrow, with wallets created with xpub containing fingerprint and derivation still works fine.

jreesun, 2022-09-12:
Are these single-key or multisig wallets?

odudex, 2022-09-12:
Single-key

jreesun, 2022-09-12:
Figured out the issue(s). 

Apparently, BlueWallet _does_ use the key origin header to look for the key's fingerprint. If it's not there, it sets the fingerprint as `00000000` in the PSBTs it generates. This means that Krux won't be able to match the fingerprint on the tx input and will refuse to sign.

Unfortunately, when I made a last-minute change to zpub QR codes to stop including key origin headers in an effort to reduce the QR code density for 22.08.0, I introduced this regression. I'm going to revert my change to zpub QR codes which will fix this.

Further, I ran into another issue when debugging this... the UR decoder library we're using will throw an exception if you are parsing a single-part UR and call `expected_parts_count`. It looks like someone else ran into this: https://github.com/Foundation-Devices/foundation-ur-py/pull/7 I'll be fixing this in that PR as well.

Sorry for the headaches! I'll cut a new hotfix release in the next day or so when I get time.

As for the user on Telegram, they will need to update to this new release, delete their wallet in BlueWallet (funds are safu), then re-import a new wallet from the (now-corrected) zpub QR code. When they refresh the wallet in BlueWallet it should fetch the old transaction history. Then when they create a new outgoing transaction, it will include the fingerprint on the PSBT and Krux will sign it.

odudex, 2022-09-12:
Nice, I have already warned the users. Thank you!

odudex, 2022-09-12:
I tested it here and apparently it is fixed. I've uploaded binaries with the fix so others in Telegram group could test it.
I also tested on Electrum, and it won't accept prefixes on xpubs and zpubs, so it never worked anyway.

jreesun, 2022-09-13:
In the meantime, here are steps for Electrum if someone wants to try with the new release:
Import zpub, then remove the key origin information in brackets, first copying the fingerprint value.
In the Console tab, add:
```
>>> this_key_index = 0 # the index of the key in the wallet, 0 for single-key
>>> this_key_fingerprint = "00000000" # the fingerprint you copied earlier
>>> this_key_derivation = "m/84h/0h/0h" # m/48h/0h/0h/2h for multisig (replace the first 0 with 1 if on testnet)
>>> wallet.get_keystores()[this_key_index].add_key_origin(derivation_prefix=this_key_derivation, root_fingerprint=this_key_fingerprint)
```

closed 2022-09-13 by jreesun

jreesun, 2022-09-13:
@odudex FYI, just published the new release with these fixes

odudex, 2022-09-13:
Great, will share the news in groups and Twitter!

PR #178 (2022-09-13): Fix zpub QR exports and single-part UR parsing
Fixes #177