selfcustody/krux #184: Live Debug don't work, device resets because of watchdog

tadeubas, 2023-02-07:
I've tried to follow the steps here to live debug a MaixDock: https://github.com/selfcustody/krux#live-debug-a-device

Everything works but after a while Krux resets the device, the image below shows the terminal running, CRTL+C pressed, some arbitrary math executed and suddenly the Krux watchdog restart:

[image]

jreesun, 2023-02-07:
Makes sense. That's unfortunately the watchdog working as designed. This can be worked around by increasing the `RESET_TIMEOUT` value in `src/wdt.py` during development and testing: https://github.com/selfcustody/krux/blob/main/src/krux/wdt.py#L24 

If this isn't sufficient, there may be a way to modify `maixpy_main.c` in the MaixPy repo to disable the watchdog on interrupt.

tadeubas, 2023-02-07:
Yes I have disabled it! You just need to add this line on the `src/wdt.py`:
```
wdt.stop()
```

The only problem is that you need to compile the code and flash the device... I will submit a PR complementing this info in README

tadeubas, 2023-02-08:
I've created a way to disable Krux watchdog without the need to build the project from source (see the open PR).

Instructions are presented at `README`, the user only needs to create a property called `WATCHDOG_DISABLE` in the `config.json`file at the device `/flash` (this is possible to do just before the Krux watchdog resets the device)

PR #186 (2023-02-18): Multiple features (watchdog disable, sd hot-plug)
### New Config to user disable watchdog to Live debug a device

Closes #184  - Added possibility to disable the Krux watchdog via config property `WATCHDOG_DISABLE`
closed 2023-02-18 by jreesun