selfcustody/krux #199: Annoying print prompts

odudex, 2023-02-28:
The thermal print prompts are really annoying for those not willing to use the printer, I think we should make the printer disabled by default before next release.

jreesun, 2023-03-01:
Confused about this. When I added the prompts, that was how they functioned--you only saw the question if it detected the printer. Did this change?

odudex, 2023-03-01:
On my understanding the only function the collect information from the printer is `has_paper`. It could be used to "detect" the printer but, it's doing this.
I don't even connect the printers Tx to Kruxs Rx because, on my understanding, it could damage the Krux hardware, as printers TX outputs 5V logic, while Krux operates at 3.3V logic.

odudex, 2023-03-01:
Update: Measured the TTL output voltage of my printer and it is 3.3V, so printers Tx could be connected to Krux Rx, I'm not sure this is the case for all printers. Not even Adafruit's printer datasheet contains TTL level, they just state: "you simply need a 3.3V-5V TTL serial output from your microcontroller to print text, barcodes, bitmap graphics, even a QR code!" This is OK to connect a "serial output" only. But it would be good idea the measure the printers output to be sure it is 3.3V before connect it to Krux input

jreesun, 2023-03-01:
Just saw @tadeubas discussion where he said:
> Created a None Printer Driver in Settings as default, in order to avoid Print? prompt to users that don't have a printer
This works for me.

odudex, 2023-03-01:
> Just saw @tadeubas discussion where he said:
> 
> > Created a None Printer Driver in Settings as default, in order to avoid Print? prompt to users that don't have a printer
> > This works for me.

For me too! Closing.

closed 2023-03-01 by odudex