selfcustody/krux #203: Feature request: inconspicuous startup.

streamofstars, 2023-04-16:
**Feature request:** inconspicuous startup e.g. by default, start the Krux installed wallet to a screen with fake error or run dummy feature (e.g. show camera output on screen) and allow access to actual Krux wallet only after specific key-press combination, e.g. press a button quickly 5 times in a row.

**More details:** It might be difficult to achieve considering this software is maybe not exactly hardware-agnostic but still, allows installation on various hardware but it would be great if the default startup behavior was, for example, to display some cryptic, fake error, or start some completely crypto-unrelated feature like e.g. for devices with camera it might start some photo taking app, etc. The actual wallet software would be accessed only after a specific key pressing sequence is executed e.g. one would have to quickly push a button 5 times or maybe start the device while holding an extra button. I don't have an exact idea in mind, the above is just a rough idea so please do not bash me for this specific example, and I definitely do not have proper knowledge on how to implement something like that but the idea is to make a device with Krux wallet installed to be inconspicuous hardware wallet. I.e. if a third party person takes the device and turns it on they must be convinced that this device is NOT a cryptocurrency hardware wallet.

** Why: ** Imagine you are being stopped and searched by police or you are crossing an international border or taking out your electronics on a tray for x-ray at the airport and it so happens that LEOs/customs officers/airport security/whatever agents are suspicious about people carrying a lot of cash or valuables with them and they also consider cryptocurrency as such. They see a Krux-installed device and ask what is that. You answer something like it's a device for experimenting with IoT, AI camera or whatever. Then they ask you to turn it on and show it to them. With the current version of the software you are busted because upon turning it on it's obvious what it is - maybe not really to literally a random person but LEOs may have some training in recognizing money laundering, money smuggling, and so on, so there is a very small but not non-existent risk they recognize Krux. Definitely now, upon turning Krux equipped device on, the story about it being an IoT/camera something will not hold longer then 3 seconds. If we have e.g. a screen with some fake, cryptic error (some memory segfault whatever) by default, then you can always say it is an IoT/camera whatever but it's broken, I'll turn it on for you, see that, it just throws this error. Of course, this will not stop well trained agents but it simply minimizes the risk of discovering it is a hardware wallet by a random person, a thief, or poorly trained government agent.

tadeubas, 2023-04-16:
Nice suggestion! We have a plan to add a game like pong, or something else, right after startup! For now we are testing some new features that are compiled and available at https://github.com/odudex/krux_binaries/

To see some of the new features available, check this discussion: https://github.com/selfcustody/krux/discussions/202

jreesun, 2023-08-18:
@streamofstars This is the issue being referred to https://github.com/selfcustody/krux/issues/77

It might be more "boring" than a game like pong or breakout, but I think @streamofstars' idea is better at accomplishing the goal of being discreet. Would also save on [precious space](https://github.com/selfcustody/krux/issues/228).

Going to close the other issue in favor of this one.

nervetrip, 2025-02-26:
I might have some free time in the near future and am very interested in tackling this task. I'm an experienced software engineer, but this would be my first foray into firmware dev. Any pointers or things I should be aware of? 

What's involved in getting a PR approved and merged? Assuming my implementation is well implemented, is this a feature this is likely to be included in a release?

My instinct is to build this feature as a configurable option in settings that would persist between boots. Any suggestions for a good setting name? I'm thinking "decoy mode" or "plausible deniability mode" or similar.

One consideration is we probably shouldn't use the Krux logo as the boot or screensaver animation when "decoy" mode is enabled. Using Google Lens or a google reverse image search of the logo links right back to the Krux project. I'm open to suggestion on what the decoy SPLASH content should contain, but my initial instinct is for it to not be a logo. Maybe it should simply contain the text "Corrupt firmware" or similar.

---

Reason for my personal interest:

I would love to see this added, sooner than later. I need absolute plausible deniability while traveling, and I love that Krux runs on devices that aren't immediately recognizable as hardware wallets. But if the device boots to a UX that is obviously crypto related, it completely undermines my goal.

I'm currently traveling internationally for the next 6-12 months and have left all my coldcard and other "recognizable" devices at home. The risk is probably minimal, but I don't want to chance a crypto-aware border agent flagging me for further inspection. Or worse -- notifying his buddies on the outside follow me and shake me down. 

I'm traveling with a Yahboom device with Krux installed and have been intending to flash the original AI vision camera firmware back onto it before crossing borders. This would give me plausible deniability. But it's really a hassle to swap back and forth between firmware. I've already visited 2 countries without flashing or having an issue, but all it takes is one time.

Originally I liked the idea of a simple game, but the limited firmware size is of concern. And it really doesn't solve plausibility, as it isn't realistic in 2025 that anyone is going to carry around a tiny little dedicated gadget to play the pong or snake game. 

So I too agree that the best solution would be to have a simple "corrupt firmware" message with a secret way to unlock Krux. Then I can claim the device is for "my work as a robotics engineer" and I have a broken build. But if there is enough firmware space, I'd be happy with any non-crypto related UX, especially if it looks like the UX is for AI vision or robotics since that's the original intent of K210 devices.

odudex, 2025-02-26:
This idea has been around for a while. @tadeubas plans to make it one of the Krux apps, which will be signed modules installable on Krux. You can join his efforts!

A good starting point is to download the source code and Poetry, then run the simulator. Test your first customizations there; when you feel they are ready, build the firmware and flash the device.

The project documentation contains everything needed to accomplish this, but we are here and in the Telegram group to help as well.

PRs will require format, lint, translations, and test coverage. They are welcome, and we all act in a positive and constructive way about them.

tadeubas, 2025-02-27:
@nervetrip get in touch in Telegram, I think it is faster to talk there. Also take a look on this draft PR https://github.com/selfcustody/krux/pull/485

The PR is a working version of Krux apps with an incomplete Nostr App. We would have to think in a secure way how to allow users to set the Krux app to run at startup.