selfcustody/krux #205: M5StickV: Navigation direction using side buttons is the other way around

streamofstars, 2023-05-25:
**Release:** 22.08.2
**Device:** M5StickV
**Problem:** IMHO navigation direction using side buttons is the other way around.
**Details:** When using side buttons to navigate the menu, pressing the button on the top of the right side of the chassis moves the selection down, and pressing the button on the bottom of the left side of the chassis moves the selection up. In my opinion this is counterintuitive and should be the other way around i.e. button on the top right should move selection up, and button on the bottom left should move selection down.

tadeubas, 2023-05-25:
Hi @streamofstars thx for the suggestion!
Maybe it is just your opinion or maybe we can't rely on a one size fits all approach... I think that the option to change this behavior on settings would be nice, what do you think?

odudex, 2023-05-25:
There's a historical reason for that. Krux once used only a single button to move, forward only, the upper right button. Thanks to a "hack" we managed to use power button on bottom left as "previous and move back/up. But I agree we could change this! But another problem would occur: It would move left with right button press, and move right with left button press😁

jreesun, 2023-08-18:
Curious to see if others have the same complaint, but (obviously) this never felt awkward to me. If it bugs enough people, might be worth adding a setting to invert the up/down buttons but I'd like to see how many would prefer this before a special case is added in.

jreesun, 2023-10-05:
Based on the lack of engagement with this issue, I'm going to close this as "Not planned." Will reopen if it comes up again.

closed 2023-10-05 by jreesun