selfcustody/krux #207: Load Mnemonic, Via Numbers, it is possible to enter a number out of the allowed range

streamofstars, 2023-05-29:
In option Load Mnemonic, Via Numbers, it is possible to enter a number out of the allowed range (>2048). Clicking go after entering e.g. 9999 takes users to the next word input.

IMHO there should be a value validation at each word to avoid situation in which a user makes a mistake at the beginning of the long seed phrase input process and only finds out about it after a few minutes of inputting at the very end.

odudex, 2023-05-29:
Could you check the tests version available [here](https://github.com/odudex/krux_binaries): https://github.com/odudex/krux_binaries ?
Only valid numbers are allowed, and there is also an "auto go" feature for "input as numbers".

streamofstars, 2023-05-29:
I'm sorry, I just came back to cancel this issue, this problem does not exist.

I have not noticed the words counter did not change upon my incorrect input. I have created a different issue https://github.com/selfcustody/krux/issues/208 IMHO there should be a clear message that what user provided is wrong and that they need to input that word again.

closed 2023-05-29 by streamofstars