selfcustody/krux #212: Error when scanning PBST QR code from Bluewallet with M5StickV

hugoender, 2023-05-31:
Followed guide on website to create wallet using Bluewallet app. When I try to send funds from the Bluewallet app to another wallet and have the M5StickV sign the PBST by scanning the QR code Bluewallet provides, I get the following error:

`Error: AttributeError(" 'NoneType' object has no attribute 'fingerprint' ",)` 

This appears in red text.

hugoender, 2023-05-31:
Just read this on the krux site:

> **Why am I unable to sign a PSBT from BlueWallet?**
> As mentioned above, some wallet software does not support the descriptor key expression format. In this case, BlueWallet will ignore the key origin and derivation info when importing the xpub to create a single-key wallet. This will result in the wrong derivation being used in BlueWallet and thus the inability to sign an outbound transaction in Krux.
> 
> Currently, the way to properly create a single-key wallet in BlueWallet is to export the second QR code that Krux displays which contains the zpub. BlueWallet can then correctly infer the derivation path when creating the wallet.

Unfortunately, the krux device didn't provide a second QR code. It only showed one (1/1).

odudex, 2023-05-31:
I believe these happened if you imported the wallet descriptor from blue. Blue won't inform the fingerprint so this error occurs. If you don't important the wallet you'll be able to sign after a warning. Or, if you use latest beta version you'll be warned of the incomplete descriptor from blue when you import the wallet, and will be able to sign.

hugoender, 2023-06-02:
I stand corrected. I saw the 1/1 at the bottom and didn't realize that the next screen was the zpub. I just thought it was going back to the xpub screen from the QR code screen.

Thank you for the detailed response. I'll leave this open because technically it's still an issue.

odudex, 2023-06-02:
In beta version QR codes subtitles now shows XPUB or ZPUB, and this error with Blue Wallet won't happen anymore. I hope we get it into official release soon.

closed 2023-09-15 by odudex