selfcustody/krux #233: After 'import flash', 'flash.erase()' does not appear work on Amigo TFT.

jdlcdl, 2023-08-28:
This issue is much longer than I initially intended.  TL;DR  it's not a concerning issue at all, I'm simply sharing.

In my quest to be able to explain every byte on my Amigo's 16MB onboard SPI Flash storage, I was digging into the 3MB of SPIFFS between 0xd00000 and the end of flash.  I found many copies of `settings.json` spread sparsely in this space, as well as some 4-byte markers (spaced 131072 bytes apart) that I assume are formatting.  I wanted to start with a clean slate.  While `os.remove()` and `os.flash_format()` work to remove files from SPIFFS, they're certainly not clearing data bytes; I didn't truly expect that they would.

I recall that in `src/krux/firmware.py` flash is written via `import flash` with a `flash.erase()` step prior to `flash.write()` [here](https://github.com/selfcustody/krux/blob/67578003d8e7a1c8c7122981156c8172845b81b8/src/krux/firmware.py#L132) on the develop_rc branch (and that it did so two lines earlier on the main branch) but when I tried to use it to erase blocks in SPIFFS space I had no luck.   I didn't know if SPIFFS was silently protecting me from direct access to the bytes owned by that filesystem, so I tried to use `flash.write()` and `flash.read()` to verify that it does not.  I did the same outside of SPIFFS and had success each time, however `flash.erase()` never worked for me anywhere.

I am assuming that `import flash` is made possible via [MaixPy/components/micropython/port/src/flash](https://github.com/selfcustody/MaixPy/tree/3ce545675093110401ea997415522a9446192f5d/components/micropython/port/src/flash) and a `w25qxx.h` header in `MaixPy/components/drivers/flash/include`.  Reviewing those informed me that I should only be using `flash.erase()` for 4096 and 65536 sized blocks at aligned addresses, as `src/krux/firmware.py` does, but I had already tried this to no avail.  I also tried other block sizes and did NOT get any exceptions, which surprised me.  Maybe my assumption at the beginning of this paragraph is wrong.

I was able to find a related discussion [here](https://github.com/selfcustody/krux/discussions/70) but it was more about resolving a compile-time error and not about whether `flash.erase()` worked or not.

Maybe `flash.erase()` works, and/or is necessary on some boards, and isn't supported on others?  I have found other `w25q*` drivers that use the same 0x20 and 0xD8 for erasing 4096 and 65536 sized blocks just as the krux header suggests, as well as some that do not even implement erasing blocks.

For my short term goal, I'm assuming that the `flash.erase()` step in `src/krux/firmware.py` is doing nothing for my amigo, and that the `flash.write()` steps are working just fine, so I have resorted to the following to erase my SPIFFS, which I paste and call from the usb console:
```python
def erase_spiffs():
    '''
    writes 0xff to entire SPIFFS because it appears flash.erase() doesn't work on Amigo TFT

    assumes `import flash` origin is selfcustody/MaixPy/components/micropython/port/src/flash
    '''

    FLASH_SIZE = 2**24
    SPIFFS_ADDR = 0xd00000
    BLOCK_SIZE = 0x10000

    import flash

    empty_buf = b'\xff' * BLOCK_SIZE
    for address in range(SPIFFS_ADDR, FLASH_SIZE, BLOCK_SIZE):
        if flash.read(address, BLOCK_SIZE) == empty_buf:
            continue

        flash.erase(address, BLOCK_SIZE)
        if flash.read(address, BLOCK_SIZE) == empty_buf:
            print('%d bytes at %s erased' % (BLOCK_SIZE, hex(address)))
            continue
 
        flash.write(address, empty_buf)
        if flash.read(address, BLOCK_SIZE) == empty_buf:
            print('%d bytes at %s written w/ 0xff' % (BLOCK_SIZE, hex(address)))
            continue
```
I've left it here in case others with different boards would like to confirm.  If the messages end with "erased", then `flash.erase()` indeed is working for some boards.

Note: This only works because SPIFFS is the very end of the 16MB flash, if SPIFFS were one byte larger, or moved one byte further, it could overflow into Kboot stage 0 and brick the Amigo until battery discharged, then would require ktool reflash.

In past discussions that were more focused around not bricking devices due to overflow [here](https://github.com/selfcustody/krux/issues/228#issuecomment-1683856037), It was opined that in regards to SPIFFS overflowing, we might just limit the number of encrypted seeds stored there.  Now that I have looked and found SPIFFS to be sparse and cluttered w/ old data, and with formatting markers throughout, I suspect that SPIFFS is respectful of its 0x300000 size and that this should not be a concern -- assuming that krux devs are respectful that SPIFFS address + size does not overrun 16MB.

odudex, 2023-08-28:
@tadeubas , I think SPIFFS is not used to save the firmware, it will only be used on the SPIFFS area, above 0xd00000

@jdlcdl 
Digging into the code, `flash.write()` also calls the erase method when necessary to raise bits values to `1` [here](https://github.com/selfcustody/MaixPy/blob/3ce545675093110401ea997415522a9446192f5d/components/drivers/flash/src/w25qxx.c#L270).
The only difference I've noticed is, when writing, it waits while dma is busy. Could this be a clue why direct `flash.erase()` is not working properly?
We have the demand to add a "wipe device" feature, probably under settings or tools. Most just want restore default settings and get rid of all mnemonics at once, and for that a simple file deletion would work. But using your code would be true wipe, and data would be in fact vanished. This is very relevant work!
@jreesun , @tadeubas what do you think?
To add a "wipe" feature under tools would be easier, because settings has its own way of working and is not built to call methods. But in other devices is something usually placed under settings...

tadeubas, 2023-08-28:
Thx @odudex for the explanation! For this wipe feature, I think under tools would be just fine.

jdlcdl, 2023-08-28:
> Digging into the code, `flash.write()` also calls the erase method when necessary to raise bits values to `1` [...]

It could be, I will certainly take a look; I had not even looked there yet.  Thank you for the pointer.

jreesun, 2023-09-30:
> We have the demand to add a "wipe device" feature, probably under settings or tools. Most just want restore default settings and get rid of all mnemonics at once, and for that a simple file deletion [...]
@jreesun , @tadeubas what do you think?
To add a "wipe" feature under tools would be easier, because settings has its own way of working and is not built to call methods. But in other devices is something usually placed under settings...

A "Restore defaults" option under Settings is a good idea and should be its own feature separate from wiping. You might have stored encrypted mnemonics on the device but accidentally changed some settings and just want to return to the defaults, in which case you wouldn't want to wipe out your mnemonics...

As for "full wipes," it's unfortunate that we can't do this by default now due to the possibility that mnemonics (or settings) might be stored on the flash. :/ My original vision for Krux was that it would be a stateless device with a readonly filesystem that you could knowingly plug hardware pieces into to gain features like mnemonic storage or settings persistence. For example, a [microSD for encrypted mnemonics](https://github.com/selfcustody/krux/issues/190) or microSD for settings with [mnemonics on dedicated USB-C devices like Yubikeys](https://github.com/selfcustody/krux/issues/120). This way, Krux can always choose the most hardcore "wipe" possible on shutdown, and you always know exactly where everything is stored. Hmm.

tadeubas, 2023-09-30:
Your "original vision for Krux" is still avaiable at releases under version [22.08.2](https://github.com/selfcustody/krux/releases/tag/v22.08.2). FOSS works that way, if someone want to change something, it changes accordingly to his view, if others like and share the same view, they change too on a specific repo. Lots of forks and versions could be available at the same time and the "original" and or "official" version or view could change with time.

Now regarding the feature, I don't see any problem to create a "restore defaults" and "full wipe" of what is connected to the device (flash/SDCard)

odudex, 2023-09-30:
Discussion, criticism  and questioning is healthy, and with planning and teamwork we can achieve higher gols than we would working independently. Also, something I learned with time is group achievements are far more satisfying than individual ones. Let's make an effort and to try to integrate our ideas and expectations the best possible way.
My take on this is that both settings and mnemonic storage are very convenient, and this convenience surpass the fact it wouldn't be stateless anymore. Actually, both modes are possible, so user is free to choose.
I think we can have separated "restore defaults" and "wipe device". When can use @jdlcdl method for a full wipe of each bit of the flash, keep the settings on ram, then save only the setting again on flash after the wipe, this way wiping only the encrypted mnemonics.

jdlcdl, 2023-09-30:
> Discussion, criticism and questioning is healthy
and
> group achievements are far more satisfying than individual ones.

"Amen" to that!  Each individual has blind-spots, in a team, they have fewer.

jreesun, 2023-09-30:
> Discussion, criticism and questioning is healthy, and with planning and teamwork we can achieve higher gols than we would working independently. Also, something I learned with time is group [...]

Agreed. Particularly about planning and discussion! ;)

You don't want a project run by a group of people who agree on everything, otherwise it's an echo chamber subject to mob rule.

To be clear, I can see why the feature is liked, and I'm not saying it should be removed. However, such a big change to the project should have had more discussion around it, particularly all the possible edge cases that would come as a result of a core assumption being violated. I'll take the blame here that I should have been more active and diligent about this before it was added. All I can say is that I honestly didn't have the time to dedicate to this project the past year. Development continued without me or my input, so my only recourse is to retroactively voice my concerns now.

Incidentally, this is an issue I have with large PRs that are not focused on one feature: it's difficult to review them adequately or individually discuss each feature contained within, especially when the features intertwine and become dependent on one another. I don't like making someone throw out their hard work, so the result is I have to rubber stamp the whole thing. It's like an omnibus bill in congress. The outcome is suboptimal. 

Going forward, I'm going to be more strict about forcing adherence to the guidelines for contributing here: https://github.com/selfcustody/krux#contributing

Anyway, the reason I don't like having a "Wipe Device" feature is that I think it should be the default behavior. And if we add it as an option, I think it will confuse a lot of users who will suddenly think, "Wait, all the times I used it before it wasn't wiping anything?" I'd rather we open a discussion about the different ways Krux is used, both as stateless and stateful, and see if we can more holistically design the UI around that.

In general, I think holistically reviewing the UI/UX of Krux every year or so would be useful to avoid adding too many options and menus and keep Krux simple and unsurprising (another core tenet). Established projects often run into the problem of local optimization due to not taking a step back and thinking from first principles how they would design the project if given a blank slate.

odudex, 2023-09-30:
So being objective, @jreesun, do you think the option to save settings and encrypted mnemonics on internal flash should be removed?
I don't demonize the use of internal flash, I even think it's safer to use than removable flash...

qlrd, 2023-09-30:
> Anyway, the reason I don't like having a "Wipe Device" feature is that I think it should be the default behavior. And if we add it as an option, I think it will confuse a lot of users who will [...]

In particular, I think we should remember what made us "Kruxers" (except @jreesun  who is the "Adamic Kruxer"). I don't have the technical knowledge of krux codes (my contribution is more in the pedagogical aspect through tools with GUIs). Therefore, I will give my opinion as a user.

I fell in love with the project due to a paradigmatic exchange: the airgapped, amnesic and easily accessible tripartition (it would be possible to argue: "It has SeedSigner and SpecterDIY", but so, where I live, it is much easier to have a Sipeed and installing software than assembling the previous ones).

And from this I concluded that this was *my demand*. On the other hand, there are other demands, including from people who have experience with another paradigms (coldcard or trezor, for example) whose seed needs a security component.

Krux's initial proposal is that even this would not be necessary. However, everything that is created is transformed. And with Krux it would be no different; there were demands in Telegram groups for the most different implementations, including seed memorization, whose security component is encryption (I personally use and like it stateless).

> So being objective, @jreesun, do you think the option to save settings and encrypted mnemonics on internal flash should be removed?

Would it be the case to have two firmware versions for each device? One stateless and one stateful? An initial setup on device that make it stateless or stateful? I don't know. But I think we should ask the majority of users whether this should remain the case or not .

odudex, 2023-10-02:
@jdlcdl I believe I fixed the `flash.erase()` method on Maixpy with [this](https://github.com/selfcustody/MaixPy/commit/c512739640461797f3140a3b371f81b17ec3fd00), on the branch "sipeed_update" of our Maixpy.
I noticed a reboot would be necessary after a "wipe", as the SPIFFS would be re-formatted after reboot.
Since you dug into this, it would be helpful if you could take a look and help me find out if everything is working properly!

jdlcdl, 2023-10-02:
Nice!

I recall that I had tested erasing 4096 as well as 65536 byte blocks, both with no luck, so that you have left the larger block unchanged will make for an effective test that your changes, and not something else in the sipeed update is the solution.  I will take a look again, thank you.

odudex, 2023-10-02:
The changes affect both sizes, there was an error with the "if" too. Please check both!

jdlcdl, 2023-10-02:
It's working, but actually, so is the larger block size erase, which I didn't expect to work at all since you didn't fix that in your latest commit.

I started by confirming that without your updated sipeed_update branch, flash.erase() still fails as it did in the past.

After booting firmware with your sipeed_update branch at c5127396:
* flash.erase(0xd00000, 1024) returns quickly and does nothing because it shouldn't, it's an invalid blocksize for erase.
* flash.erase(0xd00000, 4096) returns quickly and indeed sets those bytes to b'\xff', WOOHOO!!!
* flash.erase(0xd00000, 65536) returns quickly and also sets those bytes to b'\xff', which I didn't expect because it appears you didn't change the `elif (int_size==65536)` branch.

I wonder if simply updating sipeed's MaixPy solves this without your changes.   I do recall seeing some w24qxx changes, but I'll admit that I didn't understand their significance.

I did a `git reset --hard 3ff9b4a` to remove your very latest change, leaving your first edit (fix flash erase method).
Trying to flash.erase an invalid block size didn't produce an error, but it also didnt do anything, as expected.  flash.erase() of a 4096 or 65536 blocksize worked just as above... indicating that your latest change didn't hurt anything but might not have been a required fix.

I did a `git reset --hard 18e1b58` to get back to your original sipeed_update branch (removed unused IDE related code) without either of your latest flash-erase fixes.   And back at square one with flash.erase() that appears to be broken.  

Summary:
I confirm that it was commit 3ff9b4a which was the fix that made the difference, at least on my amigo.  Your latest commit doesn't break anything, but maybe is not needed?  Does this mean that it was about a type error between a micropython object and a 4byte c integer?  Anyways, great wizardry! Bravo!

---

My final test session from the usb console below after booting only your first fix w/ MaixPy at 3ff9b4a:

```
KeyboardInterrupt: 
MicroPython v1.11 on 2023-10-02; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>> 
>>> from machine import WDT;WDT().stop()
>>> from Maix import utils # needed for my all_bytes_are() function
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== def all_bytes_are(byte, begin, length, block_size=2**12, verbose=False):
===     '''
===     returns True if all bytes in flash are the same as byte, otherwise False
=== 
===     assumes that utils.flash_read() behaves as if imported from Maix
===   ==         print("\nthe %s bytes at %s are %s 0x%s." % (
===             length, hex(begin), 'ALL' if answer else 'NOT all', hexlify(byte).decode()))
=== 
===     return answer
=== 
>>> spiffs, xff, invalid_bs, small_bs, big_bs = 0xd00000, b'\xff', 1234, 4096, 65536
>>> all_bytes_are(xff, spiffs, invalid_bs)
False
>>> import flash
>>> flash.erase(spiffs, invalid_bs)
>>> all_bytes_are(xff, spiffs, invalid_bs)
False
>>> all_bytes_are(xff, spiffs, small_bs)
False
>>> flash.erase(spiffs, small_bs)
>>> all_bytes_are(xff, spiffs, small_bs)
True
>>> all_bytes_are(xff, spiffs, big_bs)
False
>>> flash.erase(spiffs, big_bs)
>>> all_bytes_are(xff, spiffs, big_bs)
True
```

and my `all_bytes_are()` function, which I cut/paste with `ctrl-E` and `ctrl-D`:

```python
def all_bytes_are(byte, begin, length, block_size=2**12, verbose=False):
    '''
    returns True if all bytes in flash are the same as byte, otherwise False

    assumes that utils.flash_read() behaves as if imported from Maix
    '''

    from binascii import hexlify

    answer = True

    if verbose:
        print("Checking if %s bytes of flash at %s are all 0x%s..." % (
            length, hex(begin), hexlify(byte).decode()), end='')

    bytes_read = 0
    while bytes_read < length:
        if bytes_read + block_size < length:
            if utils.flash_read(begin+bytes_read, block_size) != byte * block_size:
                answer = False
                break
            bytes_read += block_size
        else:
            if utils.flash_read(begin+bytes_read, length-bytes_read) != byte * (length - bytes_read):
                answer = False
                break
            bytes_read += length - bytes_read

        if verbose:
            print('.', end='')

    if verbose:
        print("\nthe %s bytes at %s are %s 0x%s." % (
            length, hex(begin), 'ALL' if answer else 'NOT all', hexlify(byte).decode()))

    return answer
```

odudex, 2023-10-02:
Maybe part of the last commit is unnecessary, but some is. If you will read a block right after erasing it, and don't wait for the `w25qxx_is_busy_dma()` the result might be wrong. The bigger error was comparing two different variable types on the if that checked the block size

odudex, 2023-10-03:
> ```python
> def erase_spiffs():
>     '''
>     writes 0xff to entire SPIFFS because it appears flash.erase() doesn't work on Amigo TFT
>
>     assumes `import flash` origin is [...]

Besides it is working I was still thinking on optimizations for this, and using your method from first post I noticed something: The `if` statement below never is `True`,  and it always erase all blocks.
`if flash.read(address, BLOCK_SIZE) == empty_buf:`
Maybe the data is too large to be compared on a single if statement?
On Maixpy C method to write the flash they check byte per byte in a for loop, if a byte needs to be erased then they break the test loop and erase the hole block.

odudex, 2023-10-03:
Also, do you agree this verification could be added directly on the Maixpy erase method (in C), as it probably would be faster than doing it within the python application?

jdlcdl, 2023-10-03:
As for the comparison never being true, I have not witnessed that in the past but I only ran this function a few times originally, then twice yesterday.  What I was intending for this code to do was to print nothing when a block is already empty (all b'\xff'), else "erased" if the flash.erase actually worked, or "written" if it had to write the bytes the slow-ish way.  If you figure out how to reproduce all bytes as b'\xff' and this still doesn't work, I'd love to be able to do it myself.  Does it work with smaller block sizes?  I do get memory allocation errors often for block sizes 128k and above, gc.collect() helps sometimes.  I've never had the same issue with 64k blocks which gc.collect() didn't resolve.

I would hope to figure out if flash.erase() is working religiously and then just assume that it will work each time.  What I've read of these SPI flash devices is that they're fast and reading and writing, but when it comes to the special erase functions, they're especially efficient (as long as there is a particular function for efficient erase, like "the whole chip" or "4k" or "64k" at a time).  I'd hate to start double-guessing and slowly verifying otherwise efficient operations that should just work as expected.

If we end up having to verify, I'd agree that in C it should be much faster, I hope it's not necessary with your latest changes.

odudex, 2023-10-03:
Wow, I've been learning a lot with your tools and methods, thank you!
I believe we can use your findings to help demystify the flash and improve the code to use it in a safe and transparent way.

jdlcdl, 2023-10-03:
Thank you for that last comment.  It makes my exploration into flash feel worth while.

I'll admit, my original thoughts when I received my amigo gift were  "Wow, 16MB of internal flash?  What's that work-out to be, like room for 500k private keys?  How can we inspect that and be sure?"

I have more hacky scripts, hopefully they'll evolve into something that others can use easily.

jdlcdl, 2023-10-04:
Remembering that I'd confirmed a reboot works to setup spiffs space after it's been erased w/ `flash.erase()`.  As well, we can use `os.flash_format()` as an alternative w/o reboot, so it's ready to receive whatever files need to be re-saved.

```
>>> import os
>>> os.flash_format()
[MAIXPY]:Spiffs Unmount.
[MAIXPY]:Spiffs Formating...
[MAIXPY]:Spiffs Format successful 
[MAIXPY]:Spiffs Mount successful 
True
>>> os.getcwd()
'/'
>>> os.listdir()
['flash']
>>>
```

tadeubas, 2024-03-14:
Can we consider this done on [release 24.03.0](https://github.com/selfcustody/krux/releases/tag/v24.03.0) ?

odudex, 2024-03-14:
Yes, @jdlcdl if there's something else please let us know and we can re-open it.

closed 2024-03-14 by odudex