selfcustody/krux #247: Translation of the error is not working

aglkm, 2023-09-18:
1. Switch to non English language
2. Encrypt and save a mnemonic into device flash storage
3. Try to load this mnemonic with the wrong key
4. Notice an error with some or all symbols not translated

tadeubas, 2023-09-18:
I tried with es-MX and pt-BR, here are the results (the only untranslated part is the type of the error). But I've noticed lots of error messages without the `t()` function, so this means that when those errors happen they will not be translated

[image]
[image]

odudex, 2023-09-18:
Fixed with c812d2ef8abddc4f6028e159981fb872fd8429e8.
Now we know it's better not to use translations when raising errors.

closed 2023-09-18 by odudex