selfcustody/krux #252: Inscribe Krux

jreesun, 2023-09-22:
Bitcoin and every full node in the world storing the firmware that can be used for self-custody of it sounds poetic to me.

It just so happens that the Krux firmware binary is around 2MB which is small enough to fit into a block as an inscription. 

However, due to the fees this would incur (~$500-$3000 per binary at current prices), it would likely be too cost prohibitive to inscribe every release.

Is there further compression we could do to drastically reduce the size of firmware?

If not, then perhaps it would suffice to do a special one-time inscription of each binary from the first Krux release that [enables taproot support](https://github.com/selfcustody/krux/issues/103) and has demonstrated itself to be stable after several months of usage.

Unfortunately, I can't do the inscribing and merely wanted to put this idea out into the wild.

tadeubas, 2023-09-22:
If this happens, it could mean that the Krux project officially supports the Ordinals thing and the use of the Bitcoin blockchain to store data. I wonder if people see this as more of a political act than a poetic one, and that could affect the image Krux has been building. Will we also support Ordinals transactions and the separation of uncommon satoshis? Many questions could be raised.

odudex, 2023-09-22:
Inscriptions disturbed my sleep for a few days when people started to flood the mempool, raise fees and inflate the UTXO set size. Maybe I was a FUD victim, but it's still not something I like. Now it's a reality off course we all must accept it.
As Bitcoin is about freedom, I wouldn't bother if someone does it, but even if I could, I wouldn't inscribe data on blockchain, and if the project had a budget, I would encourage to prioritize other demands.
I'm afraid the impact for this action, if done by project members, would be a love and hate (probably more hate) engagement. It would be an unnecessary controversy for the project.
The Taproot(also legacy, nested segwit and custom derivation) support, however, is something we could prioritize.

jreesun, 2023-09-30:
It's unfortunate that using Bitcoin to give people around the world a permanent, immutable, uncensorable place to download firmware for self-custody of it could be viewed negatively by proponents of Bitcoin. 

I don't know of a safer, more permanent place to store untampered downloads of Krux than Bitcoin itself. BitTorrent is a close second, but seeders need to have a copy and be online for others to download it. Also, and more importantly, they have no economic incentive to continue seeding Krux into the future, especially if it becomes "difficult" to do so. With inscriptions, you pay miners a fee to store the data, and the file will last forever.

Sidestepping the arguments for and against ordinals and inscriptions as a whole, it's a fact that they exist right now... Therefore, in my mind, why not take advantage of them while they're here? Who knows... maybe Microsoft won't be so kind to continue hosting downloads of self-custody software in the future.

tadeubas, 2023-09-30:
Like you pointed out, there's always another way to share files across the internet... if M$ didn't like the project and remove from Github, we could use other alternatives like https://bountsr.org/nostr-based-github/

kkdao, 2023-10-10:
> It's unfortunate that using Bitcoin to give people around the world a permanent, immutable, uncensorable place to download firmware for self-custody of it could be viewed negatively by proponents of [...]

Indeed it is a reality we all must accept now whether we agree with it or not - I didn't agree with it at first. Since it is a reality, I think it would be a good idea to take advantage of it in the way that you proposed. I can see Bitcoiners getting behind using their full nodes for that. In case Microsoft does decide to do that, and also I think it would set our project apart from others. So coming from a marketing perspective, I think it's good.

That being said this brings up a whole other discussion, which is where would the funds come from to do these occasional inscriptions? Correct me if I'm wrong, but I don't believe we've discussed $ yet. So since @odudex mentioned prioritizing funds, however they were raised, I would also be curious to see a comparison because maybe they could be used more efficiently in the meantime. 

I hope this comment helps, because after reading this thread, it seemed it needed another opinion (for others readers to compare to) and to bring up some questions so we can all come to an agreement/understanding in the end! Also it's not dev related, so this is one way I can tell myself I'm contributing...lol