selfcustody/krux #254: USB no longer being recognized by the computer

edbfraga, 2023-09-27:
I'm a newbie.

Why my MaixAmigo is no longer being recognized by my computer?
I'd like flash another firmware for tests.

tadeubas, 2023-09-28:
Are you using the correct port on amigo? See this: https://selfcustody.github.io/krux/faq/#why-isnt-my-amigo-device-being-recognized-when-connected-to-the-computer-usb

Also, try another port on your PC and even another USB cable

edbfraga, 2023-09-28:
I already install Krux firmware, after that my USB plug doesn’t recognize my MaixAmigo as a device.

odudex, 2023-09-28:
Please give us more details, like, what's you OS? Is your Krux working? How to you check if the computer is detecting?

edbfraga, 2023-09-28:
1. I’m using Windows 10.
2. My Krux is working fine. 
3. I opened the Device Manager.

edbfraga, 2023-09-28:
I already try with three different cable.

edbfraga, 2023-09-28:
I tried again. And now this is working.

closed 2023-09-28 by edbfraga

jreesun, 2023-09-30:
Did you build the firmware from source inside a Vagrant VM and flash that way @edbfraga ? 

If so, your host computer won't recognize any USB connections from that device while the VM is turned on since VirtualBox "steals" the USB connection to pass it through to the VM guest. You can only talk to the device via the VM while it's on.

You can workaround this by running `vagrant halt` to stop the VM thus freeing up the USB port, or manually shut down the VM in VirtualBox. A full restart of your computer would also do the trick.

That's my best guess for what you experienced and my recommendation if you experience any future problems.