selfcustody/krux #258: Update embit version to latest (0.7.0)

jreesun, 2023-10-04:
It's been a while since our embit dependency was updated. Here's the latest tagged release:
https://github.com/diybitcoinhardware/embit/releases/tag/v0.7.0

Another reason to update is that I'd like to soon (™️) update the PSBT signing code to [use the PSBTView](https://github.com/selfcustody/krux/issues/2) which should solve OOM errors for large transactions (many inputs/signers).

odudex, 2023-10-09:
Apparently it's a smooth update, the only thing that should be updated together are descriptor receive/change pair notation, from `{0,1}` to `<0;1>` .

Not related to Embit but I don't know exactly why, I started to have issues on tests parsing `UR_OUTPUT_MULTISIG_WALLET_DATA` descriptors.
Tracked the error to `module 'urtypes' has no attribute 'crypto'`
And solved by changing `urtypes` with [this](https://github.com/selfcustody/urtypes/pull/6)

tadeubas, 2023-10-09:
I saw the commit, it is just an import, test passing now, for me it is ok!

odudex, 2023-10-09:
Yes, available to test on [integrated_changes](https://github.com/selfcustody/krux/tree/integrated_changes)

tadeubas, 2024-03-14:
Can we consider this done on [release 24.03.0](https://github.com/selfcustody/krux/releases/tag/v24.03.0) ?

odudex, 2024-03-14:
Yes, done in [Release v24.03.0](https://github.com/selfcustody/krux/releases/tag/v24.03.0)

closed 2024-03-14 by odudex