selfcustody/krux #259: Increase UR memory efficiency

jreesun, 2023-10-05:
Do some profiling on the [urtypes library](https://github.com/selfcustody/urtypes) and the [UR library](https://github.com/selfcustody/foundation-ur-py/tree/8308a4a6399cc361a3caaef92f0924ddc0d7e102) from Foundation Devices to see how we can improve RAM usage

odudex, 2023-11-24:
Did two things that reduced resources usage and increased speed a lot, without changing UR
1. Used binary images (1 bit per pixel) instead os "01" strings (8 bits per pixel) to store and handle QR codes. b56a33f85842a67348c98ac70d11a00e56634c76 (and some subsequent commits)

2. Inspecting both Krux and UR I figured out Krux would was calculating QR code sizes through iterations, starting with huge QR codes, and encoding them on each iteration. So I replaced the iterations by equations. de99bfe9c692068cc9262c8773101b4e01d8aa44

Learned that UR not efficient on information density, and is terrible for small screens. But let's try to use it the best way possible.

tadeubas, 2024-03-14:
Can we considered this done on [release 24.03.0](https://github.com/selfcustody/krux/releases/tag/v24.03.0) ?

odudex, 2024-03-14:
Yes, done in [Release v24.03.0](https://github.com/selfcustody/krux/releases/tag/v24.03.0)

closed 2024-03-14 by odudex