selfcustody/krux #268: Remove Logging functionality

tadeubas, 2023-10-11:
Log (INFO, DEBUG, ERROR) is something for the developer and not for the end user. As a programmer, I have never used the functionality.

There are other easier ways to debug and many features don't have any logs (which would have to be created).

When a user reported a problem, shared a photo or video or described the problem, we never requested log files (I do not consider requesting a file from the user to be an appropriate approach when dealing with an HW).

We have a memory problem in the code, we cannot have much loaded, especially code that is not used.

I vote to remove this unused functionality - https://selfcustody.github.io/krux/getting-started/settings/#logging

odudex, 2023-10-11:
I agree. We remove and clean it up.

tadeubas, 2024-03-14:
Done on [release 24.03.0](https://github.com/selfcustody/krux/releases/tag/v24.03.0)

closed 2024-03-14 by tadeubas