selfcustody/krux #278: Isolate screensaver related code

odudex, 2023-10-26:
For better code maintenance, resources usage estimation and control, screensaver related code should be moved from `context.py`, `input.py` and `__init__.py` to a dedicated file, from which a single method should be called from outside(ideally). It could also re-use already available methods for animations (ex in animated QR codes).

jdlcdl, 2023-10-31:
Also to consider, while it's on my mind (but I don't have a solution is mind), are the amounts of getter() calls to check the screensaver setting whenever waiting for user to press a button... both while a menu is being displayed, as well as when the screensaver is running.  The code in question is around...
https://github.com/selfcustody/krux/blob/846bd0e0eb3c283a7036cabbbde7b90453e0ec43/src/krux/input.py#L209

I noticed this while I was playing with (adding print statements to) krux.settings.Store.get() and krux.settings.Store.set().

As I mentioned, I don't have a solution, but it caught my eye so I thought @odudex or @tadeubas might want to take a look, in case it can be optimized.  Count me into the discussions and/or testing as you get ideas.

tadeubas, 2023-10-31:
I don't know if we need to optimize this part of the code, as the checks occur many times per second and the K210 will be running at full speed (and is capable of doing lots and lots of calculations)... but we could change this to a variable before the loop , and just check the variable there

jdlcdl, 2023-11-01:
The getters and setters are super fast when just in memory.

tadeubas, 2024-01-23:
On Maixpy_dock version 24.04beta14 the isolated screensaver code has a bug, it cannot appear on the screen that has an infobox like: `Home > Extended Public Key > XPUB - Text`. I've noticed a little blink on the screen when the code tries to start screensaver but nothing happens

odudex, 2024-01-23:
It's hard coded to not load if there's an info box. I thought it was a goal when you created it, not showing up whenever there was information on screen. Would you like to change this and load it independently of screen?

tadeubas, 2024-01-23:
No, I think it is fine! Thx for clarifying it to me.

odudex, 2024-01-24:
Done with 9b13f48f2efb1a7089f624a9236a61bdfdd4217e

closed 2024-01-24 by odudex