selfcustody/krux #281: Feature request: Calculate last (checksum) word when giving 11 or 23 words.

smartsourcefreak, 2023-10-28:
In the tools section  - new topic "Calculate last word":

When chosen, new screen: "Enter 11 words" and "Enter 23 words".
When chosen, new screen: Enter as "Words", "Word Numbers".
When choosen let user input an incomplete Mnemonic (11 or 23 depending on the choice above).
When 11 respectively 23 words/numbers were entered again show the entered words/numbers as indexed list where field 12 (or 24) is blank or ???? and ask to Continue to calculate ALL word/numbers options for the "?????" field, Continue "Yes - No (Back to Main Menu).
If chosen "Yes" show an indexed list with all valid last words/numbers. At the bottom ask the user to choose his word/number. After user did: New screen "Choose word / number: WXYZ as 12th (/24th word)? "Yes - No (Back)".
If he choose "Yes" show the complete indexed Mnemonic. Ask user to "Load and work with Mnemonic?" or "Dicard/Back to Main Menu"
If user chooses "Load and work with the Mnemonic" continue as in "Load Mnemonic" where there's asked for BIP39 passphrase (Type/scan/No bip39) and so forth (show fingerprint, single/multi - loading).

Would be nice to have a "Calculate last word" feature like this or similar.

odudex, 2023-10-28:
The feature exists, when you are `Loading a Mnemonic`, just press `GO` when in word 12 or 24, with input as blank. It will choose a word automatically. What could be enhanced is: Instead of randomly choosing one, show a list of valid so user can choose, problem is for 12 words the list would be too big (128 words)

smartsourcefreak, 2023-10-28:
Oh. Regrettably that's quite hidden. No one would ever think, krux could do it. To be honest, I like my way of having it implemented much more as it's obvious at all that the feature exists. Yeah, a (touchable) list of valid words would also be great! Regarding the 128 possibilities in the 12 words setup; why not show 24 words at page 1 and let the user switch between six pages as a whole of vaild indexed words? That would be cool! Like choosing page 2 in a google search, having the words 25 to 28...with left and right arrows on the bottom to do the navigation. But maybe it's too difficult to implement. On the other hand the page-switching method could also be useful in future changes/improvements.

odudex, 2023-10-28:
Yes, this feature, and some others are hidden like Easter eggs 😁. It's worth checking the docs: [Getting started](https://selfcustody.github.io/krux/)
It's hard to make information more abundant while keeping the compatibility with tiny screen devices,  but we're trying. Thank you for the feedback.

tadeubas, 2023-10-29:
If you omit the 12 word or the 24 (just press Go) Krux will suggest one last word based on the number of times you pressed a button, if you don't like the word suggested, just don't accept the last word and repeat the process, Krux will change it every time to a valid last (checksum) word.