selfcustody/krux #283: [minor issue] Index button of the previous addresses is not correct?

aglkm, 2023-11-15:
Shouldn't we need to correct the index of the previous addresses?

Change the following:

`def list_address_type(self, addr_type=0):`
...
`"%d..%d" % (num_checked - max_addresses, num_checked),`

to:

`"%d..%d" % (num_checked - max_addresses + 1, num_checked),`

[image]

jdlcdl, 2023-11-15:
Just here to chime in that "Index starts at 0... while Numbers start at 1".  I'm sort of making that up... but just for consistency and terminology, "index" sounds computer-ish so that I remember it starts at 0, and "number" sounds human so that we start counting at 1.  It's growing on me that the first address (with a derivation path that ends in /0) is "1."

... and I agree with @aglkm above!

odudex, 2023-11-15:
Good catch! Wouldn't you like to make a PR?
I think we should keep the numbers, starting at 1, as we also refer to as first, second address etc. and just make the change @aglkm suggested.

aglkm, 2023-11-15:
Sure. I will add a PR tomorrow.

aglkm, 2023-11-16:
Code changes added here https://github.com/odudex/krux/pull/36

closed 2023-11-16 by aglkm