selfcustody/krux #297: [Bug] passphrase QR with apostrophes doesn't scan correctly

2oh1, 2023-11-30:
When scanning a passphrase QR that contains an apostrophe, the apostrophe gets deleted and replaced by the preceding letter.

For example, in a passphrase QR, "they've" scans as "theyyve", which leads to an incorrect wallet (xpub/zpub/etc).

This issue doesn't occur when typing the passphrase manually.  In other words, when typed manually, "they've" is shown and processed correctly as "they've".

I'm running Krux v. 23.09.0 on a Maix Amigo.

Also, just to confirm (since it's a tiny symbol), the ' shown below the # and to the left of the ( is an apostrophe, correct?  Just checking, though I'm sure it is.

tadeubas, 2023-11-30:
Hi @2oh1 , regarding the characters available to select, they are listed below (yes, it shows an `'` apostrophe below the `#` on the screen):
```
LETTERS = "abcdefghijklmnopqrstuvwxyz"
UPPERCASE_LETTERS = "ABCDEFGHIJKLMNOPQRSTUVWXYZ"
NUM_SPECIAL_1 = "0123456789 !#$%&'()*"
NUM_SPECIAL_2 = '+,-./:;<=>?@[\\]^_"{|}~'
```

odudex, 2023-11-30:
> [They're / Theyyre](https://imgur.com/a/YAQPeXI)

The QR code contains "They’re" not "They're". Do you see the difference? Krux doesn't have the ’ glyph, only '.

2oh1, 2023-11-30:
Interesting!  I had to really zoom the screen to see the difference, but I do.  I typed it with an apostrophe on a Mac.

odudex, 2023-11-30:
Yes, you did nothing wrong! Good you make these tests and helped to find this potential source of trouble.

closed 2023-12-03 by odudex