selfcustody/krux #299: Better inform the user the risks of using encrypted mnemonic stored on flash or SD card

tadeubas, 2023-12-04:
As of `23.09.0` Krux allows users to store an encrypted mnemonic on flash or SD card. It is also possible to delete the stored mnemonic. But as @jdlcdl investigations pointed out https://github.com/selfcustody/krux/discussions/269 the delete word may pass the wrong idea to the user. Users need to be aware that deleting a mnemonic stored on flash will not remove completely the mnemonic from flash, but will only mark the content as deleted and an attacker can obtain that mnemonic later with physical access to the device. It is important to notice that this issue may affect the storage on SD card too.

I know we can add more info on the [docs ](https://selfcustody.github.io/krux/getting-started/usage/navigating-the-main-menu/#encrypt-mnemonic) but my point is that we can't rely only on the docs to inform the user, we need to add a warn message every time the user will store or delete a mnemonic from flash or SD storage. This is specially important because the user may test this functionality with their real mnemonic and a poor key (just to see how encrypted mnemonic works), then later delete the mnemonic hoping everything is ok and the device or the SD card are clean. As @jdlcdl pointed to me "delete really isn't delete" and we need to make that clear to the user, tell them what Krux does (and maybe what they should do).

odudex, 2023-12-05:
As Krux features grow it will get harder to forecast every possibility of usage and how users understand the inner workings and the risks of the tools they use. We must aim for a balanced set of instructions and restrictions. Too much and them and the UX will be compromised, to little and we may have risks and bad usage materializing. We will not be able to teach about filesystems through M5stickV screen, so it is important to be objective and minimalist at the device's UI, while keeping details in documentation. I believe the "delete" word is still appropriate, as it does the same thing as it does in any OS, where there's no warning someone could retrieve that information using memory sweeping tools.
Having that said, i think we could:
- Add a message when deleting from SD, ex:
  "To ensure data is not recoverable, full format your SD card using a computer."
  On documentation we better describe what a full format is and when it is appropriated to do it.
 - Add a message when deleting from flash, ex:
  "To ensure data is not recoverable, use Wipe Device feature from Settings"
  On documentation we better describe wipe feature.
- Move `Wipe Device` feature from `Settings` to `Tools`, to be together with `Delete Mnemonic`

tadeubas, 2023-12-05:
I would suggest some additions / changes:
* On the screen `Home > Encrypt Mnemonic`:
  * Store on SD:
    "Warning: Krux cannot completely remove the data after this operation. Proceed?"
  * When the key inserted has less than an "adequate amount" (an arbitrary threshold) of entropy:
    "Warning: Storing your mnemonic with this weak key is dangerous. Continue?"

* On the screen `Tools > Delete Mnemonic`:    
  * Delete from SD:
    "Warning: Krux can't guarantee that this data will not be recoverable after this operation."
  * Delete from Flash:
    "Warning: For complete data removal, use Wipe device."

jdlcdl, 2023-12-05:
I like the above ideas of better messaging.  I'm not sure about exact wording.

In tight screen space, I wonder if a softer verb than "delete", like remove (or drop, unindex, unlink), might hint that it's not destructive delete; elsewhere where is, stronger verbs like: wipe, secure delete, scrub will be seen and considered, then used whenever appropriate.  Same in different languages as translators know best (seems challenging).  Maybe there are icons that can express the nuance.

jdlcdl, 2023-12-06:
tl;dr on a gist I'll share below:

It appears that the an "msdos" (as in `mkdosfs`) filesystem on a "W95 FAT32" MBR partition does NOT treat overwriting a file similar to how it happens in spiffs.  Rather, on the sdcard when I alter settings.json or seeds.json, it truly feels like 0x00 bytes are overwritten on top of bytes that were deleted... and appending doesn't just abandon a copy and write the entire file into a new page like on spiffs.

Similar to spiffs, a delete (as in `rm filename`) on the sdcard filesystem is effectively unlinking the filename, abandoning the bytes which are left on the sdcard.

More research is warranted, but mnemonics that are on sdcards and later deleted seem to be less of a concern than I originally thought.  My tests do not exhaust what happens when files get fragmented and scattered, so Im not sure how long this convenient pattern plays out.  I'd still side towards "learn how to `dd if=/dev/zero of=/dev/sdcard` on your computer!"

https://gist.github.com/jdlcdl/2939dec9bf2dc5bcaf1dac1801f3c21d

odudex, 2023-12-06:
Another jaw dropping investigation @jdlcdl! Even with some lost bytes left behind, this is much better than spiffs and than I expected.

closed 2024-03-14 by tadeubas