selfcustody/krux #307: [Bug] Fail from wake up from sleep on random Maix Amigos

odudex, 2024-01-14:
When user press shutdown on Maix Amigo, to turn it on again, a single click on power button should be necessary.
However I have a device, and had other reports, where this is not the case. The device won't wake up after entering sleep mode. To turn it on again it is necessary to hold the power button for more than 6 seconds, to shut it down, than press the power button again to turn it on.
I don't know exactly what's causing this, I've spent hours trying to figure out if there was a register on the AXP173 pmu to set that would fix it, but I failed.
I also tried to exchange the M2 "brain/CPU" modules from two Amigos, one with, other without the problem. And I concluded the problem is located at the main board ("motherboard"), not at the "brain board". The AXP173 and power button circuitry are located at the main board, so it is probably something there, and hopefully a register at the AXP173 the current firmware is not setting that is causing this.
The fact that there are no AXP173 English datasheet makes the problem harder to solve, but even after reading the available AXP192(similar model) datasheet to look for clues, I couldn't find any.
Sipeed won't answer any technical question too.

jdlcdl, 2024-01-14:
I confirm having similar on my amigo_tft purchased around June 2023.  I'll be happy to help test ideas, or to provide details about my amigo in hopes that a solution can be found.

odudex, 2024-01-24:
Fixed with 5530a5640cdbddcc01e0c46049d09bf7fdf4bc97

closed 2024-01-24 by odudex