selfcustody/krux #308: [Bug] Settings enter flash_message sometimes appear and sometimes not [maixpy_dock - 24.04.beta14]

tadeubas, 2024-01-23:
When entering `Settings menu` a flash_message is shown. But on version 24.04.beta14, sometimes it is shown very briefly or it doesn't appear at all!

I've tested entering the `About menu` to check if it was a double click issue, but not a single time a single click was interpreted as a double click (entering the about screen and exiting), only a real double click would enter and immediately exit the About screen. Is this an error too? a fast double click is NOT respecting the `debounce` value, but when you click and rotate almost immediately it ignores the second input, respecting the `debounce`. Don't know if it is related to this issue or not, plz tell me if I need to open a separate issue for that.

tadeubas, 2024-01-23:
Updating the issue:
It appears to be related to the input on Maixpy Dock. When I selected `New Mnemonic` > `Via Camera` > `12 words` > `Proceed? Yes` It showed a message `TOUCH or ENTER to capture` and then took the photo right away... Only one time I was able to really take the photo, every time it appears that a click was performed right after the message `TOUCH or ENTER to capture` dissapears

odudex, 2024-01-24:
Fixed with c49132231b07296b64334804537e0ec4bc6cd234

closed 2024-01-24 by odudex