selfcustody/krux #337: [Enhancement] Security option in Settings for hide mnemonic and timeout shutdown

tadeubas, 2024-03-21:
Some users want to be more secure when loading a wallet by not showing the mnemonic and encrypting the mnemonic menus after loading. Others want an automatic shutdown after some inactivity.

So the idea is to create another menu entry in **Settings** called **Security** and add these two entries:

- Hide mnemonic
- Timeout shutdown

When enabled, hide mnemonic will **hide** the **Mnemonic** and **Encrypt Mnemonic** entries from the **Main Menu** (only for loaded wallets, not for newly created ones).

Timeout shutdown allows the user set a time in minutes from 0 (disabled) to 60. It should shutdown the device in any screen if idle for the configured time.

yoonseokham, 2024-03-24:
Hi, I'm using Krux as my main wallet and I love it TYSM! 
I really want this feature you suggest. also, It would be great to have an option in the settings that allows enabling or disabling the feature to ask for a password before PSBT signing

tadeubas, 2024-03-24:
> Hi, I'm using Krux as my main wallet and I love it TYSM! I really want this feature you suggest. also, It would be great to have an option in the settings that allows enabling or disabling to ask for a password before PSBT signing

Plz elaborate more on how we would register / save this password to check when singning a PSBT?

yoonseokham, 2024-04-17:
sorry for the late reply.  I think it would be good enough for the idle timeout shutdown.

odudex, 2024-04-17:
Auto shutdown timer setting is now available on beta version.