selfcustody/krux #349: [Bug] Cube some pages don't use the full screen

tadeubas, 2024-04-02:
Some pages uses only half of the screen on Cube, like it was running on a M5stickV

**Device**
- [ ] Amigo
- [ ] M5stickV
- [ ] Dock
- [ ] Bit
- [ ] Yahboom
- [X] Cube

**Version**
- [ ] Official release (selfcustody/krux): 
- [X] Beta release (odudex/krux_binaries):  

#### Describe the bug

On cube or simulator:

- `Load Mnemonic > Via Manual Input > Tiny Seed (Bits) > 12 words` (uses only half of the screen)

[image]

- `Load Mnemonic > Via Manual Input > Tiny Seed (Bits)` (could increase size usage like in other devices)

[image]

- `Backup Mnemonic > Other formats > Stackbit 1248` (doesn't show the extra info like other larger devices)

[image]

- `Backup Mnemonic > Other formats > Tiny Seed` (uses only half of the screen)

[image]

odudex, 2024-04-02:
I see the enhancements opportunities but, for the sake of priority, what exactly is the bug?

tadeubas, 2024-04-02:
Lol, I will change it to Enhancement 😉

odudex, 2024-05-08:
Done

closed 2024-05-08 by odudex