selfcustody/krux #354: [Enhancement] Save a list of receive/change addresses to SD card, or print!

tadeubas, 2024-04-07:
Some users like to have a list of their receive/change addresses to put in a spreadsheet or to check/use them more easily. Others may want to print.

The idea is to have an option in the address explorer to, based on a number entered by the user, save the next (50?) addresses on the SD card in CSV format. Another option would be to print th next (5?) addresses.

jdlcdl, 2024-04-08:
This is a great idea!

CSV or even pre-formatted text -- one address per line, would work great for me.
If there are fields, which ones?  For receive and change addresses, or only for receive?

```
BIP32 master fingerprint: 01ab89ef
Derivation path: m/84'/0'/0'

Receive Addresses:
../0/35: bc1q...
../0/36: bc1q...
../0/37: bc1q...
...

Change Addresses:
../1/35: bc1q...
../1/36: bc1q...
../1/37: bc1q...
...
```

Note: I didn't include the actual xpub, because maybe user is willing to be sloppy about leaking a few addresses w/o losing all past/future addresses (and why would they even be printing addresses if they wanted to be capable of creating all of them?)

tadeubas, 2024-04-08:
Yes for receive and change, but only address would be exported, no additional info