selfcustody/krux #373: [Enhancement] Draft: Simplified UI/UX option

tadeubas, 2024-04-27:
Draft version, we would need to check with these users what is actually needed...

### Description
As we add more features, some screens may contain too much information or small buttons for people who are visually or physically impaired.

The idea is to add an option in the settings called “Simple UI”. When enabled, it would simplify/personalize some screens to help these people interact with Krux, allowing them to be sovereign people.

DINOPACON, 2024-04-27:
Isso é maravilhoso, eu mesmo gostaria muito de uma interface mais limpa como por exemplo:

Criar novo mnemônico 

Carregar mnemônico

Opções avançadas

Configurações 

Assim, quem quisesse utilizo opções avançadas ou fora do padrão de carregar uma wallet ou criar uma, tivesse essa possibilidade.

Mas seja como for, é uma iniciativa maravilhosa.

tadeubas, 2024-07-27:
One idea that came to mind is to enable sequential steps instead of multiple options using menus. Sequential steps asking one think at a time may be better to people that feel overwhelmed with information.

odudex, 2024-07-27:
The multiple hardware compatibility provided by Krux introduces an extra layer of complexity in terms of UI design and maintenance. In my view, incorporating multiple user interfaces would introduce yet another layer of complexity. This would lead to a substantial increase in conditional coding, testing, and documentation, which would, in turn, greatly affect the speed and quality of development.
While it is a noble goal, I wouldn't want to discourage it. Instead, I would recommend creating a fork. Perhaps a "Krux Lite" version where alternative concepts such as "sequential workflows" could be explored. A fork would also permit someone to take on the responsibility of maintaining the code, tests, and documentation, potentially leading to more streamlined codebases for both the main and "Lite" versions.

jdlcdl, 2024-07-28:
Just to chime-in with my first thought on this:

It's a simple "?" or "help" call to action button that is always available (much like "< Back").  It's not for in-device help, rather it's for an in-device qrcode that directs the user to the appropriate online documentation.

Implemented, it might mean that each "page" has a single unique keyword that helps build the qrcode's url.  And as the online documentation changes/reorganizes, the "help-via-qrcode" library would re-build these links so that an up-to-date device never has help-via-qrcodes that point to broken links.

As a hazard, we might be teaching users the unsafe habit of scanning qrcodes from device to their phone... but this is already the case (ie: extended public keys, signed psbts and messages, etc) so perhaps there is a solution for the only dangerous time that a qrcode should NOT be scanned with an online device: Mnemonic Backups -- which might have some visual clue that this is a "secrets" (perhaps a red margin around the screen).

tadeubas, 2024-07-28:
Nice suggestion @odudex ! And nice new feature @jdlcdl ! I think this "?" QR code help could be added to this issue: #347