selfcustody/krux #375: [Enhancement] Support SLIP-39 Shamir Backup

htprww, 2024-04-30:
Trezor supports the SLIP-39 Shamir Backup, which allows the m-of-n backup:
https://github.com/satoshilabs/slips/blob/master/slip-0039.md
https://trezor.io/learn/a/what-is-shamir-backup

I would like to see krux to support creating and importing the Shamir backups, which would reduce the risk that the Trezor company get out of business and the users have no way to restore their seeds.

tadeubas, 2024-05-01:
plz see this discussion first: https://github.com/selfcustody/krux/discussions/267

htprww, 2024-05-08:
Based on my understanding, multisig approach is suitable for multiple persons managing the wallet in collaboration, Shamir Backup is ideal for individual cold storage.
If we think about the single person cold storage use case, the "Single point of failure" mentioned in that article is not an issue, because the standard backup has the same problem. You need to construct the key on a single device for both standard and shamir cases.
The real benefit of Shamir backup is that you could separate your seed words in multiple places, e.g., a 2-3 setup could be one physical copy at home, one physical in office, one digital copy encrypted in your hard drive or dropbox. It will protect you from losing one of seed words. And it is very unlikely that someone could steal your physical and digital copy at the same time.

tadeubas, 2024-05-08:
To separate the seed words in 2, for example, I would recommend [SeedXOR](https://seedxor.com/). Shamir doesn't have a standard implementation and does not have broad support like BIP39 and multisig already have. Also, if you would like to see support for SeedXOR in Krux, plz open another issue

htprww, 2024-05-08:
Thanks for the update.