selfcustody/krux #377: env for poetry

j-moreno-c-r, 2024-05-01:
a proposal of solution for the problem of  the pip installation poetry, in the README.md, the solution is, run the "pip install poetry" inside a python venv....

j-moreno-c-r, 2024-05-01:
There may be a problem at this stage when trying to directly install Poetry, we have a solution proposal(this comands are for the unix system mac or linux):
In a local directory run(we consider the name of the venv .krux but you can change this...):
```bash
python -m venv .krux
source .krux/bin/activate
```
The result will be something like:
```bash
(.krux) username:~/directory name$ 
```
Now you can run normaly the pip of the poetry:
```bash
pip install poetry
poetry install
```

tadeubas, 2024-05-01:
plz create a PR to branch `wallet_customization`: https://github.com/selfcustody/krux/pulls

j-moreno-c-r, 2024-05-01:
ok

odudex, 2024-05-08:
Instructions for this workaround added to docs.

closed 2024-05-08 by odudex