selfcustody/krux #380: [Enhancement] allow re-type seed words if checksum not valid

htprww, 2024-05-03:
#### Describe the new feature or enhancement

I got into this issue that when I restore the wallet using seed words, I have a typo in one of the words, and did not pass the final checksum. Then the krux gets back to the Restore wallet main menu, so I have to re-type all.

I would like to stay in the seed words typing page, and allow the users to navigate back the word list, so I can just fix the typo for the single word.

tadeubas, 2024-05-03:
But how would you fix the first word and keep everything else? Isn't it a restart of the process in general?

I was thinking about this, but it would only work if you made a typo in just the last word

jdlcdl, 2024-05-03:
would it be like a multi-choice menu, one option per word... move up or down then select (or touch) to change that word (which would be defaulted filled out so would have to del the chars, retype them and go -- which will jump to the mnemonic_is_valid() check?

htprww, 2024-05-03:
I think we need the "back" and "forward" button, so the user could navigate back to see the previous entered words. 
The already entered words will be automatically filled in when you navigate back.

So if I see the invalid checksum at the last step, I will probably just need to navigate back to check the words I have already typed in, and only need to modify the typos.

odudex, 2024-05-03:
It's important to note that it is impossible to make typos; Krux will only allow valid words at each step.

Garrysoon, 2024-06-07:
> It's important to note that it is impossible to make typos; Krux will only allow valid words at each step.

the typo could be in the whole word

jdlcdl, 2024-06-07:
Imagining an important use case on this, rather than "sloppy data entry":

User did the best they could to backup their mnemonic on metal, but still it was partially destroyed.  They're missing letters from 2 adjacent words, and they'll need to try and try until they fall on a valid mnemonic with a familiar passphrase-less fingerprint.

I can see a few places where it would be useful to be able to back-out and edit words (or word numbers).

A) They can see the final list of words that they are about to accept, but they have not yet accepted... could back out here if they see a problem.
B) They've accepted the final list, but it's not a valid mnemonic... should be able to edit here.
C) They've accepted and the mnemonic is valid.  The passphrase-less bip32 fingerprint should be visible here, so that they can back out and edit words if its not familiar.

Instead of going "prev" and "next" inside a keyboard, I think the list should be visible in one "edit mnemonic" screen, "prev", "next" to highlight different words and "select" to jump into the keyboard to edit that one -- then "Go" would bring them back to the "edit mnemonic" screen.

tadeubas, 2025-02-26:
Hey @htprww  I think this issue was resolved with release [v24.11.0 ](https://github.com/selfcustody/krux/releases/tag/v24.11.0).

Plz close this issue if it is resolved, or update / open another with what is still missing. Thx!

tadeubas, 2025-04-14:
@odudex I think this issue can be closed too.

closed 2025-04-14 by odudex