selfcustody/krux #384: [Enhancement] support SeedXOR in krux

htprww, 2024-05-08:
Currently the ColdCard supports SeedXOR https://seedxor.com/. 
I would like to see it in krux, so we could generate and restore seedxor backups, and cross-check with the coldcard's implementation.

tadeubas, 2024-05-11:
Nice, the SeedXOR would be useful to create a new mnemonic using two different sources of entropy. Generating a mnemonic [via camera](https://selfcustody.github.io/krux/getting-started/usage/generating-a-mnemonic/#camera) and [via dice](https://selfcustody.github.io/krux/getting-started/usage/generating-a-mnemonic/#dice-rolls) then combining both to be your new mnemonic.

As Stepan said [in this article](https://medium.com/cryptoadvance/hardware-wallets-can-be-hacked-but-this-is-fine-a6156bbd199):
"generating the private keys on a compromised device is a bad idea, so we should use our own source of entropy instead. We can use dices, coins or any other source of entropy. The best way is to use multiple entropy sources and XOR their outputs"

jdlcdl, 2024-05-13:
I'm wondering which other software and hardware wallets support SeedXOR.  As far as I know, it's only cold-card presently (though I've seen requests for SeedXOR elsewhere in general).

SeedXOR is also a solution for separating a seed backup into more than one secret while the seed backup has not yet been leaked but the user would rather have multiple shares so that they can destroy the original backup before it's leaked.  Krux already has a solution for this, it is the encrypted mnemonic backup.

As devil's advocate, I'd like to raise the possibility that maybe coldcard could support krux's encrypted mnemonic backup (to peer-review it and/or improve it).
An argument against that is that SeedXOR is very easy to do mostly by hand needing only a standard hww or secure bip39 tool to calculate the final words... having an implementation in more hww wallets does seem natural IF this solution takes off and is wanted by more users.

GregTonoski, 2024-09-28:
There is the software BIP39-XOR: https://github.com/GregTonoski/BIP39-XOR

Sami32, 2025-09-24:
https://docs.rs/crate/seedxor/latest/source/README.md
https://gist.github.com/0300dbdd1b/568ef21fe605234b0ec00efb90759a33