selfcustody/krux #399: [Enhancement] Inclusion of special characters as accents in the letters for pashphrases

brunohitman, 2024-05-28:
#### Inclusion of special characters as accents in the letters for pashphrases like [á,é,í,ó,ú,â,ê,î,ô,û...]

jdlcdl, 2024-05-29:
Just a thought on this:
* when creating a new bip39 passphrase... device should be strict and only allow keys that are most common to most keyboards, else it could be difficult to recover on different devices/wallets.
* when recovering an existing bip39 passphrase... device should be tolerant in allowing as many keyboards as might have been used to create the original wallet.
Not sure how best to achieve that on a stateless device, but would love to hear others' ideas on this.

tadeubas, 2024-06-08:
I've tested this with Sparrow, [Seed Tool](https://bitcoiner.guide/seed/) and Krux create a different fingerprint when scanning `á` as passphrase from QR code. This is a problem even if Krux can't type these special characters, both Sparrow and Seed Tool show `411905a1` for fingerprint, but Krux show `9325504a`

brunohitman, 2024-07-09:
This is a big problem!

odudex, 2024-07-09:
Incorporating a variety of language glyphs could compromise the user experience and pose risks; therefore, we currently have no plans to implement such a feature.

earthdiver, 2025-11-03:
> I've tested this with Sparrow, [Seed Tool](https://bitcoiner.guide/seed/) and Krux create a different fingerprint when scanning `á` as passphrase from QR code. This is a problem even if Krux can't [...]

Isn’t this because Krux wasn’t fully compliant with the BIP-39 specification (UTF-8 NFKD)? It needs to perform normalization to NFKD
-- for example: passphrase = unicodedata.normalize("NFKD", passphrase).

In release 25.10.1, it seems the only hardware wallet supporting BIP-39–compliant passphrases that allow UTF-8 may now be CryptoGuide’s heretical fork. Perhaps it wasn’t necessary to insist on strict ASCII -- UTF-8 would likely have been fine (for example, the probability that a random 16-byte sequence is recognized as valid ASCII or UTF-8 is on the order of 10^-5 and 10^-4, respectively; even the latter is sufficiently small, and the probability drops rapidly toward zero as the data size increases).

odudex, 2025-11-12:
> My suggestion: if we detect **non-ASCII characters** in a passphrase (which can only happen when using QR codes), we should **warn the user** that their passphrase might not produce the same fingerprint in other wallets, since [full BIP39 compliance requires NFKD normalization](https://github.com/bitcoin/bips/blob/master/bip-0039.mediawiki#from-mnemonic-to-seed).

I'm ok with the warning idea

jdlcdl, 2025-11-12:
I agree with the above example.

Note: embit.bip39.mnemonic_to_seed() accepts its mnemonic and passphrase arguments as "str".  When an english wordslist is used then there will be no confusions turning the mnemonic to bytes (which pbkdf2_hmac() needs), but whatever the current rule is (I believe there is no normalization and simply str.encode("UTF-8") is used (certainly this is the only encoding available in micropython)), then it might already be the case that the user wants this, even if normalization would alter the passphrase bytes, especially if they were able to create their wallet with non-ASCII characters and it was embit that created it.

brunohitman, 2025-11-12:
That makes perfect sense, and I fully agree that full Unicode normalization (NFKD) is heavy for Krux.
However, I think it would still be valuable to implement at least a minimal normalization layer for common Latin-based characters — such as accented vowels (á, à, â, ã, é, ê, í, ó, ô, ú), the cedilla (ç), and basic punctuation symbols like ?, !, ., #, *, etc.

These cover most cases used in Latin languages and would already make the system much more practical for a large portion of users, without requiring the full Unicode database.

I also agree that showing a warning for unsupported characters is a good idea — perhaps the message could clarify that only the most common Latin diacritics are supported at the moment due to memory limitations.
That would strike a nice balance between usability and compliance.