selfcustody/krux #401: [Enhancement] making manual changes to the seedphrase, automatically generated by the KRUX camera

Garrysoon, 2024-06-07:
very useful to add human entropy to seedphrase, automatically generated by KRUX camera.

tadeubas, 2024-06-07:
This will be possible with this issue: https://github.com/selfcustody/krux/issues/384

Garrysoon, 2024-06-07:
> This will be possible with this issue: #384

perhaps seedXOR allows this, but simply being able to arbitrarily make changes to a ready-made seedphrase is much easier to implement.

tadeubas, 2024-06-07:
Making changes to a non-verifiably generated mnemonic will not guarantee the verifiability of the process and the user will continue to trust Krux code. The only verifiable option would be to generate the mnemonic from data and combine it with the unverifiable mnemonic generated by Krux via camera. Making changes to an unverifiable process will not make the process better.

Garrysoon, 2024-06-07:
> Making changes to a non-verifiably generated mnemonic will not guarantee the verifiability of the process and the user will continue to trust Krux code. The only verifiable option would be to generate the mnemonic from dice and combine it with the unverifiable mnemonic generated by Krux via camera. Making changes to an unverifiable process will not make the process better.

In my opinion making changes to an unverifiable process will MAKE the process better... 
At least I know that these are individual words generated by me + trust to the creator of original seed generator from a picture with good entropy = a partially verifiable process. Much better.
In addition, such human changes are available in anover generators.
+ plus, it can limit this possibility only for a part of the words, for example, up to 12 out of 24, and then even entering all 12 identical words will not make the seed phrase more than just 12 words.

odudex, 2024-06-08:
Independently of adding this feature or not, with a little patience you can already can do it in Krux:
1. Create entropy with camera.
2. Note down the mnemonic words.
3. Create new mnemonic from words
4. Use the camera mnemonic as reference, change the words you want. New algorithm will allow only a valid checksum new last word.

Or similarly with tinyseed:
Export camera entropy as tinyseed 
Go to load from tinyseed 
Toggle some bits differently from the original. 
Valid checksum will be calculated in real time.

Garrysoon, 2024-06-09:
> Independently of adding this feature or not, with a little patience you can already can do it in Krux:
>
> 1. Create entropy with camera.
> 2. Note down the mnemonic words.
> 3. Create new mnemonic from [...]
This opportunity is wonderful.  But we continue to strive for the perfection of the ideal Swiss knife.  In any case, thanks to Odudex and the team.

odudex, 2024-10-11:
Done in release 24.09.0

closed 2024-10-11 by odudex