selfcustody/krux #404: [Enhancement] need a pin to log into krux, setting limits for a one-time transaction,

Garrysoon, 2024-06-12:
1. need a pin to log into Krux.
2. possibility of customization MAX limits amounts for a one-time transaction.
3. the ability to separate work with a seedphrase and work with signing transactions using this seedphrase. (for the ability to delegate sending bitcoins to someone without revealing the seed phrase itself)

jdlcdl, 2024-06-13:
regarding the items above:  I'll make some assumption and ask for you to confirm if they are correct or need clarification:

1.  The mnemonic (either encrypted or not) and the passphrase (entered at keyboard or via qrcode) already act to perform this functionality.  Without decrypting a mnemonic in flash or sdcard with a "key" that can be as strong or as weak as the user wants... krux is already logged out.  Loading a mnemonic and/passphrase is the same as logging in.  I sort of see the "key" to an onboard mnemonic as the pin in this example, which is much more secure than a 4-digit pin would be (as long as the user choose a stronger key).

2. I understand the reason for MAX limits on spending a transaction, as it's been discussed in the tg group.

3. I need clarification on this part.  Are you intending to hand the krux device to a trusted other, like an employee, so that they can sign transactions?  So you would log them in by loading a key and passphrase, then they could sign anything that's already spendable in the wallet, but perhaps not greater than a max limit?  In the next release, there is already the possibility to "hide" mnemonics, and once entered into krux, the passphrase is not available for viewing on the screen... and since getting back to settings would require a reboot...  I can understand that a trusted other might be able to spend without learning the mnemonic or passphrase that the owner might have entered.  If they don't reset krux and leave it powered on... this could last as long as it has power and is not reset.

Garrysoon, 2024-11-04:
> Что касается вышеперечисленных пунктов: я сделаю некоторые предположения и попрошу вас подтвердить, верны ли они или требуют уточнения:

> 1. Мнемоническое правило (зашифрованное или нет) и кодовая фраза (введённая с клавиатуры или с помощью QR-кода) уже выполняют эту функцию. Без расшифровки мнемонического правила во флэш-памяти или на SD-карте с помощью «ключа», который может быть настолько надёжным или ненадёжным, насколько пожелает пользователь... krux уже выйдет из системы. Загрузка мнемонического правила и/или кодовой фразы — это то же самое, что вход в систему. В этом примере я рассматриваю «ключ» к встроенной мнемонической системе как пин-код, который гораздо надёжнее, чем четырёхзначный пин-код (при условии, что пользователь выберет более надёжный ключ).
> 2. Я понимаю, почему существуют ограничения на максимальную сумму транзакции, так как это обсуждалось в группе tg.
> 3. Мне нужно уточнить этот момент. Вы собираетесь передать устройство Krux доверенному лицу, например, сотруднику или супругу, чтобы они могли подписывать транзакции? То есть владелец будет входить в систему, загружая ключ и кодовую фразу, а затем они смогут подписывать всё, что уже можно потратить в кошельке, но, возможно, не более установленного лимита? В следующем выпуске уже будет возможность «скрывать» мнемонические фразы, и после ввода в krux парольная фраза не будет отображаться на экране, чтобы пользователи могли подписываться, не зная секретов владельца. Поскольку для возврата к настройкам потребуется перезагрузка, они также не смогут отключить функцию «скрытия мнемонической фразы».

> regarding the items above: I'll make some assumption and ask for you to confirm if they are correct or need clarification:
> 
> 1. The mnemonic (either encrypted or not) and the passphrase (entered at keyboard or via qrcode) already act to perform this functionality.  Without decrypting a mnemonic in flash or sdcard with a "key" that can be as strong or as weak as the user wants... krux is already logged out.  Loading a mnemonic and/passphrase is the same as logging in.  I sort of see the "key" to an onboard mnemonic as the pin in this example, which is much more secure than a 4-digit pin would be (as long as the user chooses a stronger key).

The fact is that the passphrase protects a separate passphrase... the PIN code protects the entire crookes firmware from being examined.... you can make this pin code easier and prohibit logging in after 3 errors... like in trezor

> 2. I understand the reason for MAX limits on spending a transaction, as it's been discussed in the tg group.

yes.

> 3. I need clarification on this part.  Are you intending to hand the krux device to a trusted other, like an employee or spouse, so that they can sign transactions?  So owner would log them in by loading a key and passphrase, then they could sign anything that's already spendable in the wallet, but perhaps not greater than a max limit?  In the next release, there is already the possibility to "hide" mnemonics, and once entered into krux, the passphrase is not available for viewing on the screen, so they could sign without learning owner secrets.  Since getting back to settings would require a reboot, they couldn't disable the "hide mnemonic" setting either.

Yes, you got it right

tadeubas, 2025-02-26:
Hey @Garrysoon 

> 1. need a pin to log into Krux.

Is this not somehow achievable with [TC Flash Hash at Boot](https://selfcustody.github.io/krux/getting-started/settings/#tc-flash-hash-at-boot) ?

> 3. the ability to separate work with a seedphrase and work with signing transactions using this seedphrase. (for the ability to delegate sending bitcoins to someone without revealing the seed phrase itself)

And this is not achievable with [Hide Mnemonics](https://selfcustody.github.io/krux/getting-started/settings/#hide-mnemonics) ?

If so, plz close this issue and open another with only one feature request (don't use one issue for multiple requests). Thx!

Garrysoon, 2025-04-18:
> Is this not somehow achievable with TC Flash Hash at Boot ?

It's a beautifully implemented feature, but pin is different

xavierb1777, 2026-03-01:
For me PIN is good because user could have stored mnemonic phrases in the flash and she wants to be inaccessible for malicious person. PIN makes stole data more difficult. Obviously it is not perfect, but more difficult means more secure. Other hardware wallets have it so it seems advisable.