selfcustody/krux #424: [Enhancement] Coin flip feature for new mnemonics

4rkad, 2024-07-17:
It would be good option to add coin flip feature to create a new mnemonic. Just 2 options to choose, heads and tails. 128 flips to 12 words mnemonic and 256 flip for 24 words mnemonic.
I know tinyseed is a similar feature but can be difficult to understand for some users. And can be difficult to do in small devices like Yahboom and others.

jdlcdl, 2024-07-18:
Would this be a way of loading a mnemonic? 
or of creating a new mnemonic (since 128 or 256 I assume you mean this), 
or might this be a "Tools" feature where user enters 11 coin flips and learns that one word (and maybe the word-number in decimal, hex, and octal)... then writes it down, in order to create a new mnemonic with the last word aided by krux?

4rkad, 2024-07-18:
Could it be in New mnemonic > via words > coin flip feature or load mnemonic > via manual input > coin flip feature
You have 2 choices, heads and tails. You flip 128 or 256 times a coin and create a walle of 12 or 24 words.

rodolfhoalves, 2024-08-02:
> Could it be in New mnemonic > via words > coin flip feature or load mnemonic > via manual input > coin flip feature You have 2 choices, heads and tails. You flip 128 or 256 times a coin and create a walle of 12 or 24 words.
> 
> Reference: https://estudiobitcoin.com/do-you-trust-your-seed-dont-generate-it-yourself/

That could be a nice feature

tadeubas, 2025-02-26:
Hey guys, this is already achievable with `Load Mnemonic > Via Manual Input > Tiny Seed (Bits)`. Choose 12w for 128 flips and 24w for 256.

Proceed as follows:
- Flip a coin, if it is heads, click on the first space (value 1) of the word, if it is tails do nothing. Repeat this step for each space up to 1024 (only up to 1024, never change 2048).
- The last word has the checksum, you will do as you did with the other words, the only difference is that you cannot set some spaces, they are calculated automatically. For 12w you will flip the coin only 7 times, for spaces 16, 32, 64, 128, 256, 512, 1024. For 24w you will flip the coin only 3 times, for spaces 256, 512, 1024.

tadeubas, 2025-04-25:
This info is now added to our docs here:

https://selfcustody.github.io/krux/getting-started/usage/loading-a-mnemonic/#tiny-seed-bits

@4rkad @rodolfhoalves do you guys think this ☝  resolves the issue?

rodolfhoalves, 2025-04-27:
For sure, thanks!